Farama-Foundation/Metaworld #4: Missing Environments

jcoreyes, 2019-09-26:
If you try running scripts/demo_sawyer.py, many of the imports don't work because of missing environment such as 
`from metaworld.envs.mujoco.sawyer_xyz.sawyer_stack import SawyerStackEnv`

zhanpenghe, 2019-10-05:
Hi @jcoreyes, thanks for reporting, those are deprecated and we will update them soon.

MishaLaskin, 2019-10-08:
@zhanpenghe the 'hello world' env of this suite - SawyerReach - is also missing from the master branch

zhanpenghe, 2019-10-08:
Hi @MishaLaskin, we use `SawyerReachPushPickPlaceEnv` for reaching, pushing and pickplace tasks.

To use it in a single task way, just do `env = SawyerReachPushPickPlaceEnv(task_type='reach')`.

MishaLaskin, 2019-10-08:
ah got it thanks @zhanpenghe! Looking forward to reading the accompanying paper when it comes out

ryanjulian, 2019-12-02:
@zhanpenghe is this fixed?

xiaofei-w, 2020-10-31:
Hi @zhanpenghe, I wonder if SawyerStackEnv is available now?

ryanjulian, 2020-11-02:
@xiaofei-w I think the current set of environments you see in the repo are everything we plan on having, at least in the forseeable future.