Farama-Foundation/Metaworld #20: AttributeError: 'MjViewer' object has no attribute 'finish'

activatedgeek, 2019-11-12:
Minimal code to reproduce the error

```python
from metaworld.envs.mujoco.sawyer_xyz.sawyer_reach_push_pick_place import SawyerReachPushPickPlaceEnv

if __name__ == "__main__":
  env = SawyerReachPushPickPlaceEnv(task_type='push', random_init=True)

  obs = env.reset()
  done = False
  while not done:
    env.render()
    next_obs, reward, done, info = env.step(env.action_space.sample())

  env.close()
```

Stack trace
```
Creating window glfw
Traceback (most recent call last):
  File "try.py", line 13, in <module>
    env.close()
  File "/miniconda3/lib/python3.7/site-packages/metaworld/envs/mujoco/mujoco_env.py", line 135, in close
    self.viewer.finish()
AttributeError: 'MjViewer' object has no attribute 'finish'
```

Gym Version: 0.12.1
Mujoco-py Version: 2.0.2.8

ryanjulian, 2019-12-02:
@activatedgeek thanks for the bug report. This is a long-standing bug in `mujoco-py` for which we need to write a workaround.

ryanjulian, 2019-12-02:
You can see we use a special helper routine for this in our test suite:
https://github.com/rlworkgroup/metaworld/blob/master/tests/helpers.py#L16

activatedgeek, 2019-12-02:
Thanks for putting in the fix!

commit b56e8db7 (2019-12-03): Fix gym.Env.close() implementation

Fixes #20
PR #29 (2019-12-03): Fix gym.Env.close() implementation
Fixes #20
commit 44b2e415 (2019-12-03): Fix gym.Env.close() implementation (#29)

Fixes #20
closed 2019-12-03 by ryanjulian