Farama-Foundation/Metaworld #22: Feature request: meta-learning baselines

Guitaricet, 2019-11-20:
It would be great to have some baselines that allow for a fast start with Metaworld (e.g. code used for the paper). Or at least to mention some repositories that use Metaworld.

ryanjulian, 2019-12-02:
@Guitaricet thanks for the FR. I agree that this would be an excellent contribution to the meta-RL community, but it's definitely out-of-scope for this repo (which intentionally includes the environments only).

We believe it's essential to separate benchmarks from implementations. If you'd like strong baseline implementations, I recommend you take a look at our sister project [garage](https://github.com/rlworkgroup/garage).

ryanjulian, 2019-12-02:
Since it's out-of-scope for this repo, I'm going to close this issue. Please don't hesitate to open a new one with a different problem.

closed 2019-12-02 by ryanjulian