Farama-Foundation/Metaworld #23: Setting environment to a reference environment's state

activatedgeek, 2019-11-23:
Here is a particular gym registered version of a metaworld environment that I am using

```python
register(
    id='MW-Reach-v0',
    entry_point='metaworld.envs.mujoco.sawyer_xyz.sawyer_reach_push_pick_place:SawyerReachPushPickPlaceEnv',
    kwargs=dict(task_type='reach', random_init=False)
)
```

In words, I'm trying to run CEM using ground truth dynamics for the reach task using fixed goal. The way I do this is by creating `n` copies of the environment and stepping on the copies. To make sure each copy starts at the current observation, I use the following code:
```python
env = gym.make('MW-Reach-v0')
# ... rollout env to some state using standard `.step()`
copies = [gym.make('MW-Reach-v0') for _ in range(n)]
cur_state = env.get_env_state()
for i in range(n):
    copies[i].set_env_state(cur_state)
```

It does reach the success state every time. Is this the correct approach? For reasons unknown to me, it does NOT work without the `.reset()` before `.set_env_state()`.

Further, now I would like to use `random_init=True` for randomized goals but the same approach doesn't work. Is it enough to just run `.set_goal_(cur_env.goal)` on each copy as below?
```python
cur_state = env.get_env_state()
for i in range(n):
    copies[i].set_goal_(cur_env.goal)
    copies[i].set_env_state(cur_state)
```

It could potentially just be a hyperparameter issue, but just wanted to make sure if I have the interaction with environments correct. Thanks!

activatedgeek, 2019-11-27:
Ping @ryanjulian. Any help here please?

ryanjulian, 2019-12-02:
@activatedgeek I apologize for the delay. Most people were away last week for the US Thanksgiving holiday. Work-life balance is important in this org, and you probably shouldn't expect much responsiveness over major holiday periods.

Please note that the API surface for the individual environments should be considered private and extremely unstable. One of the compromises we made while publishing this work was to limit the public API surface to the pre-baked benchmark configurations you see in the examples (i.e. only `metaworld.benchmarks` should be considered public).

@tianheyu927 and @zhanpenghe can probably comment better on the logic used for goals.

activatedgeek, 2019-12-03:
> Most people were away last week for the US Thanksgiving holiday....

Apologies! I completely forgot about that.

> Please note that the API surface for the individual environments should be considered private and extremely unstable....
>  .....pre-baked benchmark configurations

I appreciate the maintenance effort here! 

I believe there's much value in additional benchmarks because this task list seems to have interesting local and global structures with enough variance. Goes without being said that I would be happy to contribute a new set of benchmarks provided I make sure that my understanding of the code is correct. :-)