Farama-Foundation/Metaworld #25: Unstable simulation while using Metaworld

NagisaZj, 2019-11-26:
Hi, I am testing my algorithm on metaworld and I am frequently getting this exception

File "/home/lthpc/anaconda3/envs/MAML/lib/python3.6/site-packages/mujoco_py/builder.py", line 359, in user_warning_raise_exception
    raise MujocoException('Got MuJoCo Warning: {}'.format(warn))
mujoco_py.builder.MujocoException: Got MuJoCo Warning: Nan, Inf or huge value in QACC at DOF 0. The simulation is unstable. Time = 0.1250.

I met this problem on ML1  'reaching' and 'pushing' envs, and my codes use the same policy structure as PEARL, so I think my codes give out legal actions.

I would greatly appreciate it if you can provide some insight about what's going wrong. Thank you.

ryanjulian, 2019-12-02:
@NagisaZj These errors can occur occasionally with some MuJoCo simulations, but you should not be seeing them with regularity. If you are, I recommend you ensure that your policy is not producing actions which exceed the `action_space` of each environment.

If your actions are all compliant with the action space, I think we will only be able to help you if you can provide us with some code which reproduces the problem exactly.

Please re-open this issue if your problem persists (preferably with some code to reproduce your issue).

closed 2019-12-02 by ryanjulian

George614, 2024-06-19:
Is this still an on-going issue? I've encountered this myself with the "window-open-v2" task. I create a list of three hand-picked tasks and doing a sequential continual learning on those tasks. The issue I had was during training on the third task, i.e. "window-open-v2". The error message says the following repeateldy:

`WARNING: Nan, Inf or huge value in CTRL at ACTUATOR 0. The simulation is unstable. Time = 0.0125.`

Then the simulation malfunctions and my loss functions get `NaN` as well. I've checked that agent's action is clipped to the range of (-1, 1). 

Thanks

reginald-mclean, 2024-06-19:
@George614 can you provide more information about the actions you are using? I am quite confident this is not an issue with Meta-World but I could be wrong. As the above information mentions, it's generally an issue with inputs to the Mujoco simulator so the actions you are applying to the environment