Farama-Foundation/Metaworld #28: Pre-trained policies for each env

andreykurenkov, 2019-12-02:
I emailed this, but also seems appropriate to post here. We want to do some research that builds on top of your excellent code. I am guessing [this gif figure](https://meta-world.github.io/figures/ml45-1080p.gif) was made using trained policies for each env. Any chance you could provide the trained policies for each individual env to aid with research that uses them? The code to generate gifs of a rollout for a would also be nice, if you can share it.

ryanjulian, 2019-12-02:
hi @andreykurenkov -- sorry I didn't respond over email due to the holiday week in the US.

In short, we don't have ready-trained policies available for each env yet. The GIF in the paper was generated using policies trained with https://github.com/rail-berkeley/softlearning.

We don't have code for autogenerating the composite GIF (it was composed by hand), but if you make some that would be a cool addition in a PR.

I'll leave this issue open, since I think it would be a worthy addition from a contributor.

andreykurenkov, 2019-12-03:
Got it, thanks! If we do end up training policies we'll try to contribute :)

ryanjulian, 2020-07-14:
@andreykurenkov we don't have pre-trained policies for each env, but we do have (soon-to-be-complete) scripted policies for each environment. Do those a satisfy your use case?

andreykurenkov, 2020-07-15:
That would be quite useful (though I no longer have a need for this, at least for now -- i'll go ahead and close the issue)

closed 2020-07-15 by andreykurenkov