Farama-Foundation/Metaworld #31: A small mistake(or not) in the observation space for multi-class-multi-task environment.

RchalYang, 2019-12-06:
It looks like that the observation-space definition of the multi-class-multi-task environment is kind of confusing (I am not sure whether it's a mistake or not).

For observation space with the obs_type is **with_goal_id** or **with_goal_and_id**, current code is useing **goal_id_low** for high in Box, and **goal_id_high** for low in Box.

```python
elif self._obs_type == 'with_goal_id' and self._fully_discretized:
    goal_id_low = np.zeros(shape=(self._n_discrete_goals,))
    goal_id_high = np.ones(shape=(self._n_discrete_goals,))
    return Box(
        high=np.concatenate([plain_high, goal_id_low,]),
        low=np.concatenate([plain_low, goal_id_high,]))
elif self._obs_type == 'with_goal_and_id' and self._fully_discretized:
    goal_id_low = np.zeros(shape=(self._n_discrete_goals,))
    goal_id_high = np.ones(shape=(self._n_discrete_goals,))
    return Box(
        high=np.concatenate([plain_high, goal_id_low, goal_high]),
        low=np.concatenate([plain_low, goal_id_high, goal_low]))
```

RchalYang, 2019-12-07:
Besides, It looks like that in **observation_space** function the goal id is before the goal, which is not the case when the observation is augmented in **_augment_observation**

```python
if self._obs_type == 'with_goal_id' or self._obs_type == 'with_goal_and_id':
    if self._obs_type == 'with_goal_and_id':
        obs = np.concatenate([obs, self.active_env._state_goal])
    task_id = self._env_discrete_index[self._task_names[self.active_task]] + (self.active_env.active_discrete_goal or 0)
    task_onehot = np.zeros(shape=(self._n_discrete_goals,), dtype=np.float32)
    task_onehot[task_id] = 1.
    obs = np.concatenate([obs, task_onehot])
```

zhanpenghe, 2019-12-10:
Thanks. This is indeed a bug. We will submit a fix soon.

ryanjulian, 2019-12-23:
@zhanpenghe did you open a PR for this?

avnishn, 2020-04-30:
@ryanjulian, @adibellathur has been working on this issue

ryanjulian, 2020-07-14:
@avnishn isn't this obviated by recent changes?

avnishn, 2020-07-17:
hi @RchalYang, since we last reviewed this issue, we have released a new API for Metaworld that we believe has fixed this issue. I'm going to close this for now, but feel free to reopen and comment if you have any more questions.

closed 2020-07-17 by avnishn