Farama-Foundation/Metaworld #35: Camera Angles

karttikeya, 2019-12-23:
Hello,

Thank you for the excellent code.

I'm trying to use the setup to collect simulated RGB data for a project but I haven't been able to find a way to set custom camera angle while rendering. All I can see is some predefined modes like `topview`. Anyway I can setup custom camera views for the tasks?

Thanks!

ryanjulian, 2019-12-23:
@karttikeya we haven't extensively tested metaworld for image-based learning yet, which is why you're finding that these features are either half-baked or lacking.

Image-based learning is a feature area we plan on improving in future versions of metaworld, but I can't give you any hard commitments on when you'll see better support for this other than "some time in (early) 2020."

It sounds like you have both the motivation and the expertise to add or improve the features you need for your project. If you do make improvements, we'd love it if you take a look at `CONTRIBUTING.md` and send us a pull request with your improvements.

**Note:**
The only public API in metaworld right now is the `metaworld.benchmarks` package. You should consider the APIs for the individual environment classes private, and thus highly unstable.

karttikeya, 2019-12-24:
I see. Okay, thanks for letting me know. 
Would be happy to put in a PR if I end up implementing this in future :)

activatedgeek, 2019-12-30:
Here's a tiny hack meanwhile. 

I modified `viewer_setup()` method in [`mujoco_env.py`](https://github.com/rlworkgroup/metaworld/blob/master/metaworld/envs/mujoco/mujoco_env.py#L77) with the following statements

```python
...

def viewer_setup(self):
  self.viewer.cam.azimuth = -20
  self.viewer.cam.elevation = -20

...
```

Gets you this perspective
[image]

melfm, 2020-01-02:
So overwriting the `viewer_setup` API lets you change perspective when you do `env.render('human')`. But when you use `env.get_image` API, it still uses the default view. Btw `env.render('rgb_array')` crashes for me for some reason. 
What's the fundamental difference between setting up rendering via accessing the `viewers` vs calling `self.sim.render` ? Is there another workaround for this?

karttikeya, 2020-01-21:
Yes, modifying the `viewer_setup()` api does the trick. In my case (headless remote server, No GLFW) to use `env.sim.render` instead of `env.render` adding the following snippet directly to create and adjust the viewer 
```
env.viewer = mujoco_py.MjRenderContextOffscreen(env.sim, -1) 
env.viewer.cam.azimuth =  205
env.viewer.cam.elevation = -170
env.viewer.cam.distance = 2.3
env.viewer.cam.lookat[0] = 1.1
env.viewer.cam.lookat[1] = 1.1 
env.viewer.cam.lookat[2] = -0.1
```
renders a nice GIF-esque angle. 

[image]

reginald-mclean, 2023-02-01:
As Meta-World ownership has been transferred from the RLWorkGroup to [Farama Foundation](https://farama.org/), I am closing this issue. If there are any questions or requests for features please join our [Discord](https://discord.gg/PfR7a79FpQ)

closed 2023-02-01 by reginald-mclean

Tejas-Deo, 2023-05-08:
> 

Can you please share the code how you managed to do env.render("rgb_array")?
Thanks!

melfm, 2023-05-10:
It's been a while since I ran this but back then I modified the `def _get_viewer(self, mode):` as mentioned above. The code is [here](https://github.com/melfm/metaworld/blob/ibit/metaworld/envs/mujoco/mujoco_env.py#L186). Not sure if this would still work with the recent versions.

XZDang13, 2024-02-03:
For anyone who need modify the camera with V2 version, here is one solution.

from gymnasium.envs.mujoco.mujoco_rendering import MujocoRenderer

DEFAULT_CAMERA_CONFIG = {
    "distance": 2,
    "azimuth": 215,
    "elevation": -20.0,
    "lookat": np.array([0, 0.5, 0.0]),
}

env.mujoco_renderer = MujocoRenderer(env1.model, env1.data, DEFAULT_CAMERA_CONFIG)

Image.fromarray(env.render())
[image]

joshuasmith44, 2024-07-12:
@XZDang13 I ran this code and it seemed to work, except that setting any of the 3 indices in lookat would actually set all indices to that value. Ex. If lookat was [0, 0, 0] and I did lookat[1] = 1, lookat would become [1, 1, 1]. Does this happen in your example?

Kallinteris-Andreas, 2024-07-12:
You should be able to do `env.make(..., default_camera_config=YOUR_CAMERA_CONFIG_HERE)`, 
not sure if it is available in the current latest release, but should be for the next one.

XZDang13, 2024-07-12:
> @XZDang13 I ran this code and it seemed to work, except that setting any of the 3 indices in lookat would actually set all indices to that value. Ex. If lookat was [0, 0, 0] and I did lookat[1] = 1, lookat would become [1, 1, 1]. Does this happen in your example?

No. I can give you a demo how to set it with a wrapper.

`
import numpy as np
from PIL import Image
import gymnasium
from gymnasium.core import Env
from gymnasium.envs.mujoco.mujoco_rendering import MujocoRenderer
from metaworld.envs import ALL_V2_ENVIRONMENTS_GOAL_OBSERVABLE

DEFAULT_CAMERA_CONFIG = {
    "distance": 1.25,
    "azimuth": 145,
    "elevation": -25.0,
    "lookat": np.array([0.0, 0.65, 0.0]),
    }

DEFAULT_SIZE=224

class CameraWrapper(gymnasium.Wrapper):
    def __init__(self, env: Env, seed:int):
        super().__init__(env)

        self.unwrapped.model.vis.global_.offwidth = DEFAULT_SIZE
        self.unwrapped.model.vis.global_.offheight = DEFAULT_SIZE
        self.unwrapped.mujoco_renderer = MujocoRenderer(env.model, env.data, DEFAULT_CAMERA_CONFIG, DEFAULT_SIZE, DEFAULT_SIZE)

        # Hack: enable random reset
        self.unwrapped._freeze_rand_vec = False
        self.unwrapped.seed(seed)

    def reset(self):
        obs, info = super().reset()
        
        return obs, info

    def step(self, action):
        next_obs, reward, done, truncate, info = self.env.step(action) 
        
        return next_obs, reward, done, truncate, info

def setup_metaworld_env(task_name:str, seed:int):
    env_cls = ALL_V2_ENVIRONMENTS_GOAL_OBSERVABLE[task_name]
    
    env = CameraWrapper(env_cls(render_mode="rgb_array"), seed)
    
    return env

env = setup_metaworld_env('button-press-v2-goal-observable', 10)

obs, _ = env.reset()
img_array = env.render()
Image.fromarray(img_array)
`
When I set the "lookat": np.array([0.0, 0.65, 0.0]), the image is like this
[image]

If I change it to np.array([0.65, 0.65, 0.0]), the image is like this
[image]