Farama-Foundation/Metaworld #36: Bug(or not) about drawer-close-v1 environment.

RchalYang, 2020-01-03:
When I'm training my multi-task policy for Multi-Task 10 benchmark, I find that drawer-close-v1 task always succeeds at the first timestep of the episode. 

I checked the implementation of the drawer-close and drawer-open environments, and I noticed that they use the same goal by default and they used the same condition to judge whether the task succeeded. I'm not sure whether this is correct, would you mind taking a look at it?

RchalYang, 2020-01-03:
Drawer-close init function 
```python
        self.init_config = {
            'obj_init_angle': np.array([0.3, ], dtype=np.float32),
            'obj_init_pos': np.array([0., 0.9, 0.04], dtype=np.float32),
            'hand_init_pos': np.array([0, 0.6, 0.2], dtype=np.float32),
        }
        self.goal = np.array([0., 0.55, 0.04])
        self.obj_init_pos = self.init_config['obj_init_pos']
        self.obj_init_angle = self.init_config['obj_init_angle']
        self.hand_init_pos = self.init_config['hand_init_pos']
```
success condition:
```
objPos = obs[3:6]

rightFinger, leftFinger = self.get_site_pos('rightEndEffector'), self.get_site_pos('leftEndEffector')
fingerCOM  =  (rightFinger + leftFinger)/2

pullGoal = self._state_goal[1]

reachDist = np.linalg.norm(objPos - fingerCOM)

pullDist = np.abs(objPos[1] - pullGoal)
```
```python
info = {'reachDist': reachDist, 'goalDist': pullDist, 'epRew' : reward, 'pickRew':None, 'success': float(pullDist <= 0.06)}
```

RchalYang, 2020-01-03:
Drawer-open init function:
```python

        self.init_config = {
            'obj_init_angle': np.array([0.3, ], dtype=np.float32),
            'obj_init_pos': np.array([0., 0.9, 0.04], dtype=np.float32),
            'hand_init_pos': np.array([0, 0.6, 0.2], dtype=np.float32),
        }

        self.goal = np.array([0., 0.55, 0.04])

        self.obj_init_pos = self.init_config['obj_init_pos']
        self.obj_init_angle = self.init_config['obj_init_angle']
        self.hand_init_pos = self.init_config['hand_init_pos']

```
success condition
```python

objPos = obs[3:6]

rightFinger, leftFinger = self.get_site_pos('rightEndEffector'), self.get_site_pos('leftEndEffector')
fingerCOM  =  (rightFinger + leftFinger)/2

pullGoal = self._state_goal

pullDist = np.abs(objPos[1] - pullGoal[1])
```
```python
info = {'reachDist': reachDist, 'goalDist': pullDist, 'epRew' : reward, 'pickRew':None, 'success': float(pullDist <= 0.08)}
```

ryanjulian, 2020-01-06:
@tianheyu927 @zhanpenghe

tianheyu927, 2020-01-07:
Thanks! That's indeed a bug. I just submitted a PR to fix it.

closed 2020-01-28 by tianheyu927