Farama-Foundation/Metaworld #38: Making Environments Serializble

michaelzhiluo, 2020-01-18:
This is an issue we encountered while trying to copy multiple environments per worker, and hopefully someone can help us resolve this bug.  

In a MAML-based setup, each worker/tasks has several workers in parallel to rollout episodes. In this setup, we use `pickle` to copy an environment to other environments for each worker.

 `set_task` takes a task index and obtains the goal via indexing into `self.discrete_goals` .After some debugging, it turns out that after pickling/copying environments, `self.discrete_goals` which is a list of 50 of goals, is different for each environment. 

We are wondering if there is a way to make this deterministic with pickling. Your help is greatly appreciated!

ryanjulian, 2020-01-24:
See https://github.com/rlworkgroup/metaworld/pull/40

ryanjulian, 2020-01-25:
@michaelzhiluo does #40 solve your problem?

michaelzhiluo, 2020-01-25:
Thank you @ryanjulian  and @naeioi for the quick fix! This is good 💯

closed 2020-01-25 by michaelzhiluo