Farama-Foundation/Metaworld #43: All metaworld environments report done=True at max time steps.

krzentner, 2020-01-27:
Although this can be convenient for on-policy algorithms, it technically makes the environment partially observable. In particular, this is known to significantly increase the sample requirements for off-policy algorithms in many cases.

Since the goal of this benchmark is to compare different meta-learning methods, it would be best to have a version of this benchmark that does not unfairly discriminate against off-policy meta-RL methods (such as PEARL).

krzentner, 2020-01-27:
This may be relevant to @katerakelly and @lywong92.

avnishn, 2020-01-28:
This would also be relevant to implementations of SAC and MT-SAC.

alversafa, 2020-02-25:
Just a comment: This can also be a problem for RL^2 as it would expect `done=True` when task is complete, not when the env reaches max time steps.

krzentner, 2020-02-26:
I also thought that might be the case. Surprisingly, no environments in metaworld report `done=True` when the task is complete / on task success. This means that as long as you aren't giving RL^2 the success signal, RL^2 *should* just learn to ignore the done signal (or learns it's an "end of episode" signal) for metaworld benchmarks.

Having said that, if you're training RL^2 by mixing together the done and success signals, then yes that should cause problems on these benchmarks. It's also not clear to me if giving access to the success signal is fair in the first place, so perhaps its best not to do that.

PR #45 (2020-03-31): Disable the done signal.
This fixes #43 in what I think is the cleanest way possible. There should probably be more discussion before we officially make a v2, but this might be of use to some people immediately.

This is definitely of interest to @avnishn and @lywong92 .
closed 2020-03-31 by ryanjulian