Farama-Foundation/Metaworld #48: metaworld only works with gym 0.15.4

iDurugkar, 2020-02-11:
gym version 0.15.5 upwards have deprecated the `tags` arguments in [environment registration](https://github.com/openai/gym/blob/a99e8d153e94f532e5728b01b02b42e076334bf7/gym/envs/registration.py#L36)

They have specified the change [here](https://github.com/openai/gym/blob/4ca60758b44b6ae9080ca47a72f52c2035a4133f/README.rst#L164)

ryanjulian, 2020-02-14:
Hi @iDurugkar 

Thanks for filing this issue.

The metaworld API doesn't make use of gym environment registration, so this should not affect functionality. Is it causing you to not be able to install metaworld?

We would happily accept a pull requests which simply deleted the contents of
https://github.com/rlworkgroup/metaworld/blob/4528f84dab357d562824dac3edf159bc68399f23/metaworld/envs/mujoco/__init__.py

It's not necessary for the benchmark to work and I think is the source of your problems.

sloumotion, 2020-02-19:
I can confirm that deleting the __init__.py solved the problem for me.

Created a corresponding pull request: https://github.com/rlworkgroup/metaworld/pull/52

iDurugkar, 2020-02-21:
Oh you beat me to it. I created a pull request #53  that just removes the `tags` field from the `metaworld/envs/mujoco/__init__.py` file.

perezjln, 2020-04-01:
Hello, is the push from @iDurugkar merged already? Thanks!

iDurugkar, 2020-04-01:
No, I think it needs one more approval before being merged

ryanjulian, 2020-04-03:
closed by #69

closed 2020-04-03 by ryanjulian