Farama-Foundation/Metaworld #49: Question regarding reward function and observation space

VincentYu68, 2020-02-13:
Hi!

I have a few questions about the reward function and observation spaces that I hope you can help with.

1) The reward function for some environments (e.g. reach) is not consistent in the paper and in the code. Should I modify the code so they are consistent, or I should just use the one in the code? The one in the code makes more sense for the reaching task.

2) What was the observation space used in the ML10 and ML45 evaluations? Earlier in the paper it mentioned that the observation is 9-dimensional consisting of hand position and two task-related positions. For ML1 it hides the goal position to be the task, so it has a 6-dimensional observation space. I'm wondering if ML10 and ML45 are the same (goal position hidden), or they can observe the true goal?

Thank you so much in advance!

tianheyu927, 2020-03-02:
Hi,

1. You could use the reward function in the code. The reward specified in the paper is a simplistic version, which basically demonstrates the idea of the design of the reward functions, while the reward function in the code is augmented with a few additional components and is the actual one used for training RL algorithms. We could revise the paper to make reward functions consistent in order to avoid further confusion.

2. ML10 and ML45 should also hide the goal positions as they are using `obs_type` = 'plain'` (see https://github.com/rlworkgroup/metaworld/blob/master/metaworld/benchmarks/ml10.py#L21).

Hope this helps!

ryanjulian, 2020-03-12:
To add more context to @tianheyu927 's answer about goal positions -- goal positions are hidden in metaworld because it's a meta-/multi-task RL benchmark, not a goal-conditioned benchmark. Providing the goal position makes task inference (one of the main challenges of meta-/MTRL) trivial.

We might yet create a goal-conditioned version.

@tianheyu927 Can you update the paper with the reward functions from the code?

ryanjulian, 2020-03-12:
I'm going to close this issue for now. Please feel free to re-open (or open a new one) if you have more questions.

closed 2020-03-12 by ryanjulian