Farama-Foundation/Metaworld #65: Questions regarding the simulator

imankgoyal, 2020-03-19:
Hello,

Awesome work! I have some questions about how the meta world simulator takes care of certain cases. 

 - How does meta-world simulate grasping? Is the object assumed to be grasped if near the arm and the grip action is taken? Is there any simulation for contact or friction while grasping? Some info about the assumptions made would be very useful. 

-  As far as I understand, the action space id 4-dimensional (3 for the location of the arm and 1 for gripper). Am I correct in my observation? Does the orientation of the arm have no impact on the environment? Some info about the inverse kinematics used could be very useful here. Also, if the pose of the robotic arm matters, how can one control it?

Best regards,
Ankit

zhanpenghe, 2020-03-19:
Hi thanks for the question!

- The grasping is simulated fully by mujoco. We did not assume the object is grasped if it is near the arm/gripper and there is simulation going on during this process.

- Yes, your observation is correct. I don't think the pose of the arm matters a lot as long as a certain pose of the gripper can be achieved. We were using mujoco for inverse kinematics. 

@tianheyu927 could you please confirm?

imankgoyal, 2020-03-19:
Hi @zhanpenghe ,
Thanks for the response. I had some quick follow-ups. 
- Since all the tasks can be solved when training in the single task setup, am I correct in assuming grasping objects (per task wise) is not very difficult in the current setup (given assumptions on object sizes, the gripper, and friction).
- I may have misworded my second question before. What I meant was how much the orientation of the gripper affects the environment as no orientation info is currently present in the observation or action space.
Thanks again!

ryanjulian, 2020-07-14:
@imankgoyal 

1) In our experience object grasping behavior (based on MuJoCo's friction simulation) is fairly robust though not perfect. Please feel free to report significant regressions or instabilities if you find them.

2) The gripper orientation is fixed in a palm-down configuration to make exploration easier, so no info on orientation is needed on the observation. All environments are solvable without changing the orientation (see the scripted policies in `metaworld.policies` for proof)

closed 2020-07-14 by ryanjulian