Farama-Foundation/Metaworld #71: Observations are incorrectly padded with zeros

avnishn, 2020-04-08:
in `MultiClassMultiTaskEnv` a use case that routinely has to be dealt with is when multiple environments have observation spaces that are of different sizes. In ML45/MT50 the observation space is (6,) and (9,) for some spaces. The observation spaces that are of shape (6,) are supposed to be padded with zeros in order to make the observation spaces that are size( 6,) size (9,).

The issue is that where that is currently done, if a static_task_id, or one hot id is appended to the observation, then the one hot id is appended to the observation before any zeros are padded onto the observation. This leads to the one-hot id being incorrectly shifted inside the returned observation depending on the observaton_space shape of the underlying environment.

ryanjulian, 2020-04-08:
I agree this isn't ideal and probably unintended, but does not constitute a bug unless it's inconsistent between iterations of the algorithm. 

Is the onehot always in the same place?

ryanjulian, 2020-07-14:
@avnishn this is closed right?

closed 2020-07-17 by avnishn