Farama-Foundation/Metaworld #75: Benchmarks must be constructed before task names can be accessed

avnishn, 2020-04-13:
Currently in order to access all of the task names that belong to a benchmark's train/test set, the benchmark and all of its environments must first be constructed.

This is wasteful in the case that one needs to construct a benchmark's environments using the from task_api. The proper behavior should be that one should be able to access the names without constructing all of the underlying environments.

haydenshively, 2020-06-01:
Can't you access all of the names from EASY_MODE_CLS_DICT, MEDIUM_MODE_CLS_DICT, and HARD_MODE_CLS_DICT? The `ML1` class actually returns these through its `available_tasks` class method.

ryanjulian, 2020-06-01:
None of those dicts are part of the public API, which consists only of "metaworld.benchmarks"

they should probably be prefixed with _ to make them private e.g. `_EASY_MODE_CLS_DICT`

haydenshively, 2020-06-01:
If all `Benchmark` subclasses had `available_tasks` methods that returned the private dicts, would that be a satisfactory fix?

ryanjulian, 2020-06-01:
See https://github.com/rlworkgroup/metaworld/pull/85/files#diff-75dbfe903b808c91e2dd503cf761b531 for the right way to fix this.

ryanjulian, 2020-07-14:
@avnishn is this still true?

closed 2020-07-14 by avnishn