Farama-Foundation/Metaworld #84: Regarding the done signal at max_path_length

Kostis-S-Z, 2020-05-14:
Hello!

I read #45 and I understand why you decided to remove it, so sorry for referencing again this issue. However, I believe that as @krzentner [said](https://github.com/rlworkgroup/metaworld/pull/45#issuecomment-594911269), there are many cases that it is helpful to have this functionality. 

I also understand the [decision](https://github.com/rlworkgroup/metaworld/pull/45#issuecomment-594959879) that whoever wants to change this, will need to implement it themselves. But if there is a lot of interest of having this functionality, would it be possible to add this as an option when creating the environment? 

In the same style that there are the arguments `sample_all, sample_goals` etc, there could be an argument that would toggle between sending `done=True` when reaching `max_path_length` or not. Personally this is not an issue as I have already implement this functionality in my code as you suggest so it would be more of a future enhancement if there is a demand for it.

Would love to hear your thoughts on it!

ryanjulian, 2020-05-14:
@Kostis-S-Z As you have probably read in the issues, we removed `done=True` on `max_path_length` because it can cause severe performance degradations in off-policy algorithms, misrepresenting their true performance.

It's also a pretty subtle and hard-to-detect bug, so we regarded it as a dangerous design choice for a benchmark which is designed to be run by authors without much attention to the details. We prefer creating a little more hassle for rare use cases over causing someone to accidentally publish inaccurate results.

I'm not in favor of adding an option, because an important feature of benchmarks is that they shouldn't have any options. In reported paper results, how are we to distinguish Metaworld-with-done-is-true-on-max-path-length from Metaworld-without-done-is-true-on-max-path-length? We are actually in the process of removing the other options you mention (`sample_all`, etc.) for exactly this reason.

Unlike on-policy algos, off-policy algorithms distinguish between episodes which terminate due to the MDP reaching a true termination condition (`done=True`), and those which terminate due to a fixed horizon (`max_path_length`). Truncated but non-terminal (`max_path_length`) episodes need to be bootstrapped to account for the time steps the agent did not experience (e.g. `Q(s, a) ~ R + discount * max_{a'} Q(s', a')` given `(s, a, r, s')`) whereas true terminal episodes (`done=True`) do not need bootstrapping (e.g. `Q(s, a) ~ R`), because in the terminal case the return `R` is known to be the true, full-horizon return.

Conflating termination `done=True` with a fixed horizon actually makes the RL problem unlearnable (theoretically speaking) for the off-policy RL algorithm, because the MDP is no longer fully-observable (it can be considered a POMDP with time as a hidden variable, because you need to know time to estimate the true expected return).

closed 2020-05-14 by ryanjulian

Kostis-S-Z, 2020-05-14:
Thank you for such a detailed answer! I see your point and it makes sense.
Keep up the good work!