Farama-Foundation/Metaworld #90: Scripted policies for each environment

ryanjulian, 2020-05-22:


haydenshively, 2020-06-02:
As far as I can tell from MuJoCo’s documentation, sites are not supposed to collide with geom objects. However, in the sawyer_reach_push_pick_place environment, it looks like the sawyer’s grabber arms are intersecting with the goal sites (the red, green, and blue dots that indicate targets). This makes it impossible for the robot to put the puck on top of the target in some situations.

ryanjulian, 2020-06-02:
If you are going to start a 50-part issue, I recommend you either add a checklist to this issue or generate a new issue for each environment. 

It is probably a bad idea to try making one giant PR with all 50 scripted policies.

haydenshively, 2020-06-02:
Environments for which testing scripts have been completed (or not):
Status | Task | Notes
:------: | :--------: | ----------
[x] | Reach | Aim for `env._state_goal` instead of `env.goal`
[x] | Push | See above; also only successful 90% of the time
[ ] | Pick Place | 
[ ] | Door Open |
[ ] | Drawer Open |
[ ] | Drawer Close |
[ ] | Button Press Top Down |
[ ] | Peg Insert Side |
[ ] | Window Open |
[ ] | Window Close |

_update this list as test scripts are completed_

ryanjulian, 2020-06-02:
Can you open a separate PR for each environment? Merging a PR with 50 scripted policies will be impossible.

ryanjulian, 2020-06-02:
Also note that your scripted policy can only use information in the observation. If accessing `env._state_goal` or `env.goal` is necessary to solve it, the environment has a problem.

haydenshively, 2020-06-02:
Yeah I can open new issues.

It looks like `env._state_goal` is returned by the `_get_obs_dict()` function but not by `_get_obs()`. Does that mean I can use it? Or am I looking in the wrong place?

ryanjulian, 2020-06-03:
You can only use `obs, r, done, info = env.step(action)`

closed 2020-07-24 by haydenshively