Farama-Foundation/Metaworld #97: Scripted Policies for ML10 Environments

avnishn, 2020-06-12:
The ML10 environments that still need scripted policies include:

- [ ] button press
- [ ] peg insert
- [ ] peg remove
- [ ] window open
- [ ] window close
- [ ] sweep
- [ ] basketball
- [ ] shelfplace
- [ ] sweep into
- [ ] lever pull

@haydenshively is handling window open and window close

ryanjulian, 2020-06-17:
I updated the list above to reflect what's on master.

avnishn, 2020-06-18:
I'm going to also leave here a check list of policies that have open pull requests:

ML10/MT10 environments:
- [ ] reach
- [ ] push
- [ ] pick-place
- [ ] door-open
- [ ] drawer-close
- [ ] button-press-topdown
- [ ] peg-insert-side
- [x] window-open
- [ ] sweep
- [ ] basketball
- [ ] drawer-open
- [ ] door-close
- [ ] shelf-place
- [ ] sweep-into
- [ ] lever-pull

closed 2020-06-28 by avnishn