Farama-Foundation/Metaworld #113: CI times out after 50 minutes

haydenshively, 2020-06-20:
With all the new scripted policies, tests take longer than ever and builds are failing because Travis times out after 50 minutes.

ryanjulian, 2020-06-22:
Can you get under the limit by reducing the number of trials you run per scripted policy? 1000 is likely overkill.

haydenshively, 2020-06-22:
Yes, but with fewer trials, there's a higher chance of having an unlucky test. I'll have to decrease the success rates slightly in order to get things to pass consistently.

ryanjulian, 2020-06-22:
Yes. That is always the trade-off with statistical testing. Realistically, the difference between 100 and 1000 trials is not super large. The sampling error grows at approximately `O(1/sqrt(n))` so 100 to 1000 is not as large a jump as it seems.

If you want to keep around the 1000-trial tests for local usage, your test suite can detect that it's in the CI using the `CI` environment variable and react accordingly, or you can create separate tests with more trials, mark them as `huge` (or something else), and then exclude that marker when running in the CI. I think approach #2 is preferable and you can see it used throughout garage.

closed 2020-06-23 by haydenshively