Farama-Foundation/Metaworld #119: Some Scripted Policies are Flaky

avnishn, 2020-06-24:
While manually running the tests, I noticed that some environments have some variance in their average success rate, especially with running scripted policies with gaussian noise.

The variance is never high ~1-5 percent. I suggest that we allow for a tolerance of 5 percent, in order to avoid flaky tests.

ryanjulian, 2020-06-24:
Simply lowering the success rate until it's not flaky is a much clearer way of specifying the test. It tells you the minimum success rate given variance.

avnishn, 2020-06-29:
Will do. Instead of lowering by 5%, we'll lower by by 1% increments

haydenshively, 2020-09-14:
We haven't had any failures recently as far as I know, so I think it's safe to close this

closed 2020-09-14 by haydenshively