Farama-Foundation/Metaworld #126: SAC results source code

perezjln, 2020-06-26:
Hello, the meta-world paper is reporting results using SAC and PPO. Unfortunately, we have hard-time reproducing the SAC (and PPO) results presented in the various settings, would it be possible to provide the SAC code of the paper in this repository or maybe pointing out to the implementation of SAC which was used? Thanks!

avnishn, 2020-06-28:
Hi @perezjln, thanks for checking out Metaworld!

The implementation of SAC that was used is available in the [rail soft-learning repository](https://github.com/rail-berkeley/softlearning). Unfortunately, I don't know the implementation of PPO that was used, as I was not an author on metaworld. The implementations of MTSAC and MTPPO that were used in the Metaworld paper are not publicly available. 

However, 

Some of the maintainers, including myself, have made implementations of the algorithms that you're looking for under a repository called [Garage](https://github.com/rlworkgroup/garage). Garage has stable implementations of the original [SAC](https://github.com/rlworkgroup/garage/blob/master/examples/torch/sac_half_cheetah_batch.py), [PPO](https://github.com/rlworkgroup/garage/blob/master/examples/torch/ppo_pendulum.py), [MTSAC](https://github.com/rlworkgroup/garage/blob/master/examples/torch/mtsac_metaworld_mt10.py), and [MTPPO](https://github.com/rlworkgroup/garage/blob/master/examples/torch/mtppo_metaworld_mt10.py).

I hope this answers your question; since this isn't a direct issue with Metaworld, I'm going to close this for now, but feel free to ask more questions if you have any!

closed 2020-06-28 by avnishn

ryanjulian, 2020-06-28:
PPO results used garage MTPPO (in TensorFlow).