Farama-Foundation/Metaworld #131: Reproducing PEARL results on ML10/45

bernwang, 2020-06-30:
I am trying to reproduce the ML10 results for Pearl from figure 8 of the Metaworld paper. Here is the learning curve I reproduced: 

[image]

My results are different from the paper's. The average success rate on training tasks are frequently zero and does not resemble a "curve". Can you give me some pointers on how to reproduce the paper results? 

Basically, I used Kate Rakelly's implementation of Pearl ([https://github.com/katerakelly/oyster](https://github.com/katerakelly/oyster)). I adapted the code to log the success rate. I'm using the same hyperparameters listed in the Metaworld paper. For meta-testing, I set the number of exploration trajectories to 10, and I take the average success rate over 10 test rollouts (set num_evals=10 in Pearl's hyperparameters). 

It would be great if you can provide some insight on how to reproduce Pearl results on ML10/45 and provide some feedback on hyperparameters and evaluation protocol.

ryanjulian, 2020-07-14:
@bernwang please retry with the new API, which allows you to use `set_task` to change when an environment's configuration changes. Managing these configuration changes is key to training and evaluating PEARL and was previously quite implicit/difficult under the new API.

I have reproduced the PEARL results using metaworld and find them quite consistent.

jp18813100494, 2020-08-17:
Sorry to bother U! I'd like to reproduce ML10/45, but I am confused about the setting or meta-train tasks and meta-test tasks. Is the total env num of ML10 10? If so, what is the total task num of different task configurations? Or meta-train tasks comprise 10 different envs and every envs comprise 5 different task configurations. The same question also arises in the meta-test task.

avnishn, 2020-08-24:
Hi @jp18813100494, 

Let me try to respond to your questions in order:
> Is the total env num of ML10 10? If so, what is the total task num of different task configurations?

Yes, the total number of training environments in ML10 is 10 environments. There are 5 testing environments. You would evaluation your pearl on these 5 environments. Each of these environments has 50 different task configurations. A task configuration consists of a unique combination of (object_initial_position, goal_position). 

This means that for ML10, there are 10 * 50 = 500 unique task configurations for the ML10 training environments, and 5 * 50 = 250 unique task configurations for the ML10 testing environments.

Did this answer your question? Feel free to @ me (avnishn) if you have any more questions, or if this didn't answer your question!

Best,
Avnish

closed 2021-03-09 by avnishn