Farama-Foundation/Metaworld #137: Joint angle

emuemuJP, 2020-07-07:
How can i get joint angle or rotation of 6DoF Sawyer robot for each step?

haydenshively, 2020-07-07:
Hi! Thanks for expressing interest in MetaWorld! All of the environments in the `sawyer_xyz` folder inherit from `SawyerMocapBase`. This class has a function called `get_env_state` which should return a tuple containing (1) the joint state and (2) the motion capture ("mocap") state. That mocap state is itself a tuple containing position and an orientation quaternion.

That said, the orientation of the end effector (the tip/fingers of the Sawyer) should never change in the MetaWorld environments. It's held constant with the quaternion being `[1, 0, 1, 0]`. It's also never passed as part of an environment's observation, so the `get_env_state` function is more of a debugging tool than anything else.

closed 2020-07-07 by ryanjulian

emuemuJP, 2020-07-08:
@haydenshively Thank you very much !

I have tried to get parameters as follows.
`env = ML1.get_train_tasks('reach-v1')
joint_state, mocap_state = env.active_env.get_env_state()

MjSimState(time=0.12500000000000006, qpos=array([ 2.08394164e+00, -5.70318419e-01, -1.19094847e+00,  1.77256338e+00,
        9.66836328e-01,  1.15366423e+00,  2.40350109e+00, -1.36425008e-04,
        1.72924933e-04, -7.63206202e-02,  6.96439282e-01,  1.99999996e-02,
        1.00000000e+00,  0.00000000e+00,  0.00000000e+00,  0.00000000e+00]), qvel=array([-5.36537248e+00, -7.20529153e-02,  6.36763108e+00, -3.74349469e+00,
        1.87573374e+00, -3.64890076e+00, -1.54358735e+00, -1.24176395e-03,
       -2.47893757e-05,  0.00000000e+00,  0.00000000e+00,  0.00000000e+00,
        0.00000000e+00,  0.00000000e+00,  0.00000000e+00]), act=None, udd_state={})
(array([[0. , 0.6, 0.2]]), array([[0.70710678, 0.        , 0.70710678, 0.        ]]))
`
I am clear about mocap state, but I can't understand joint_state well.
Are there any document about joint_state?
or I would appreciate it if you could tell me more detail.