Farama-Foundation/Metaworld #145: Difference between `goal` and `_state_goal`

shagunsodhani, 2020-07-11:
Hello :)

Thank for open-sourcing the code for MetaWorld. Could you please tell me the difference between `goal` and `_state_goal` attrs of the `env` object? Since `_state_goal` is returned as part of `obs` and is used for computing the rewards, I am assuming that it is the goal that the learning agent is trying to reach. I could not understand what is the significance of the `goal` attribute. Since it is returned as part of `info`, I am assuming it has some significance :)

krzentner, 2020-07-12:
Hi Shagun,

Thanks for using metaworld. The relationship between `goal` and `_state_goal` is different in different environments. Generally speaking, the `goal` field cannot be relied upon to mean anything specific, even though it is returned in `state_info`. In older (pre-publication) versions of metaworld, the `goal` field was what `_state_goal` is now. However, it does not vary across tasks set in the same environment (even though the true goal location / reward function does).

The field `goal` field will probably be removed in the future, because its presence is confusing.

For multi-task benchmarks (MT10 and MT50), the `_state_goal` field is part of the observation, as you've noted. However, for meta-learning benchmarks (ML10 and ML45), the meta-learner is expected to adapt to the goal location without direct access to it. In other words, reading the `_state_goal` field violates the benchmark, and results that use it should not be presented as results on the ML10 or ML45 benchmark.

Hopefully that answers your question. Feel free to leave further questions if you have them.

shagunsodhani, 2020-07-12:
Thank you so much for explaining this :)

closed 2020-07-12 by shagunsodhani