Farama-Foundation/Metaworld #156: get_env_state function

emuemuJP, 2020-07-15:
_Originally posted by @haydenshively in https://github.com/rlworkgroup/metaworld/issues/137#issuecomment-655071547_

I have tried to get parameters as follows.
```
env = ML1.get_train_tasks('reach-v1')
joint_state, mocap_state = env.active_env.get_env_state()

MjSimState(time=0.12500000000000006, qpos=array([ 2.08394164e+00, -5.70318419e-01, -1.19094847e+00,  1.77256338e+00,
        9.66836328e-01,  1.15366423e+00,  2.40350109e+00, -1.36425008e-04,
        1.72924933e-04, -7.63206202e-02,  6.96439282e-01,  1.99999996e-02,
        1.00000000e+00,  0.00000000e+00,  0.00000000e+00,  0.00000000e+00]), qvel=array([-5.36537248e+00, -7.20529153e-02,  6.36763108e+00, -3.74349469e+00,
        1.87573374e+00, -3.64890076e+00, -1.54358735e+00, -1.24176395e-03,
       -2.47893757e-05,  0.00000000e+00,  0.00000000e+00,  0.00000000e+00,
        0.00000000e+00,  0.00000000e+00,  0.00000000e+00]), act=None, udd_state={})
(array([[0. , 0.6, 0.2]]), array([[0.70710678, 0.        , 0.70710678, 0.        ]]))
```
I am clear about mocap state, but I can't understand joint_state well.
Are there any document about joint_state?
or I would appreciate it if you could tell me more detail.

ryanjulian, 2020-07-15:
@emuemuJP `get_env_state()` is not a public API in metaworld, which means we don't make great effort to document it and it can change arbitrarily at any time. You should expect it to be made private or disappear in the near future, as we rewrite the API to be clear about what's public (see #141 ).

The public API for observations is exposed via `gym.Env`, i.e. the return values from `reset()` and `step()`.

The contents of `get_env_state()` is the raw state of the underlying MuJoCo simulation, and its elements can be quite different between different environments. If you would like to reverse-engineer the exact assignment from array elements to simulation joints and bodies, I recommend you read the underlying MuJoCo XML model associated with that environment and cross-reference with the [MuJoCo XML documentation](http://mujoco.org/book/XMLreference.html).

closed 2020-07-15 by ryanjulian

emuemuJP, 2020-07-16:
Thank you for your advice despite the private API！