Farama-Foundation/Metaworld #170: Rendering error

AlexanderKoch-Koch, 2020-07-20:
I have problems running a simple example with rendering. I am running this on my local machine with a display.

Here is the code
```
import metaworld
import random

print(metaworld.ML1.ENV_NAMES)  # Check out the available environments

ml1 = metaworld.ML1('pick-place-v1') # Construct the benchmark, sampling tasks

env = ml1.train_classes['pick-place-v1']()  # Create an environment with task `pick_place`
task = random.choice(ml1.train_tasks)
env.set_task(task)  # Set task

obs = env.reset()  # Reset environment
a = env.action_space.sample()  # Sample an action
obs, reward, done, info = env.step(a)  # Step the environoment with the sampled random action
env.render()
```

Here is the output
```
Creating window glfw
X Error of failed request:  BadWindow (invalid Window parameter)
  Major opcode of failed request:  40 (X_TranslateCoords)
  Resource id in failed request:  0x100a2a0
  Serial number of failed request:  218
  Current serial number in output stream:  218
```

ryanjulian, 2020-07-20:
@AlexanderKoch-Koch Thanks for trying metaworld! We would love to help.

Are you able to visualize MuJoCo-based environments from openai/gym? e.g. 
```py
import gym

env = gym.make('Ant-v2')
env.reset()
env.render()
```

Can you tell us some more details about your development environment?

1. Python version
1. Environment manager (virtualenv, pipenv, conda, poetry, none)
1. Operating system
1. Whether or not you are using Docker
1. Display drivers/GPU (NVIDIA, Intel, AMD, others?)

AlexanderKoch-Koch, 2020-07-21:
I had tried a few MuJoCo envs before. They have all worked (Reacher-v2, FetchReach-v1, InvertedPendulum). However, Ant-v2 and e.g. FetchPush-v1 almost never work. It seems to be random. I tried Ant-v2 about 20 times and launched succesfully only once. The FetchPush-v1 has about a 50% success rate.

Python version: 3.8.2
Environment manger: none
Operating System: Ubuntu 20.04
Docker: no
Intel drivers (no external GPU)

I have found a partially working workaround. I can remove the glfw.poll_events() call in MjViewerBasisc in mjviewer.py in mujoco_py. It renders successfully, but it's not interactive anymore.

ryanjulian, 2020-07-21:
mujoco_py is only tested on Python <=3.7. Can you trying using a different version of Python (e.g. by making use of `pyenv`)?

mujoco_py builds bindings to native system libraries (e.g. glfw and your graphics drivers) and unfortunately new bugs pop up every time those get upgraded. I'm not sure the extent to which it's been tested on Ubuntu 20.04. 

Because you can't get rendering to work with openai gym either, this points to a problem with your MuJoco/glfw/graphics drivers rather than a bug in metaworld.

Have you tried the standard LD_LIBRARY_PATH amendments for mujoco? e.g. `export LD_LIBRARY_PATH=$USER/.mujoco/mujoco200_linux/bin` ?

AlexanderKoch-Koch, 2020-07-21:
I have tested it now with Python3.5 and Python3.6. Now all openai gym envs work fine. However metaworld is still throwing an error.

I have found a known issue in glfw here: https://github.com/glfw/glfw/issues/1633
I was using the last release 3.3.2 from January. Now I have compiled the current master branch myself. And it's working now.

Thank you for your help.

closed 2020-07-21 by AlexanderKoch-Koch