Farama-Foundation/Metaworld #208: Why is len(ml1.test_tasks) = 50 and not 10?

Shushman, 2020-08-19:
This might be a trivial question, but I thought each ML1 environment had 50 training and 10 test tasks? (Also I wish I could tag this issue as a question rather than an actual issue)

ryanjulian, 2020-08-19:
@avnishn

avnishn, 2020-08-20:
Hi @Shushman,
This is a good question thank you for asking it!
ML1 is a benchmark where we train and test algorithms on a single environment with 50 different goal variations. For that reason, any environment that is tested in the ml1 benchmark will have the following properties, which you might have found:

`len(ml1.train_tasks) == 50`

`len(ml1.test_tasks) == 50`

Shushman, 2020-08-20:
Thanks @avnishn  I do understand your stated motivation of both `train_tasks` and `test_tasks` being 50. My query came from the paper (Pg 6; para 1)
" ML1 uses single Meta-World Tasks, with the meta-training “tasks” corresponding to 50 random initial object and goal positions, and meta-testing on 10 held- out positions".
That made me think `length(ml1.test_tasks)` should be 10, but perhaps I misunderstood?

avnishn, 2021-03-09:
closed via #312

closed 2021-03-09 by avnishn