Farama-Foundation/Metaworld #224: All the tasks are non-Markovian

hartikainen, 2020-09-10:
The original MetaWorld paper evaluates the benchmarks with standard memoryless agents like SAC, TRPO, and PPO, none of which have memory in order to solve non-Markovian tasks. However, most (if not all) of the environments are non-Markovian since they only include position information of the end effector, object, and the goal. Is this intentional? This seems to unnecessarily complicate the benchmarks and indeed with a bit more information (e.g. hand/object velocities and object orientations) the environments become solvable much more reliably.

avnishn, 2020-09-10:
Hi @hartikainen,

Thanks for bringing this up.

This is certainly a valid feedback and I addressed your point in #226.

While Github is a valid place to be having these discussions, if you're looking for more direct communication with the original authors of the paper, please join our slack community, as many of them are more active on there. You can find the link to sign up on our README.

Lastly I'm curious to know what all information did you add to the observation space in the process of doing your own rewrite of the reward functions?

Best,
Avnish Narayan

hartikainen, 2020-09-10:
Thanks @avnishn, this was useful.

> Lastly I'm curious to know what all information did you add to the observation space in the process of doing your own rewrite of the reward functions?

I simply added end effector and object velocities and the object orientation as detailed here: https://github.com/hartikainen/metaworld/blob/reward-tweaks-rebase/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py#L304-L319. I think it might be useful to also include some acceleration or contact information but I did not test things that far.

avnishn, 2021-03-09:
closed via #312

closed 2021-03-09 by avnishn