Farama-Foundation/Metaworld #225: How to create two indentical benchmark instances?

hartikainen, 2020-09-10:
It seems like the MT10/ML10 benchmarks (maybe others as well) initialize the goal positions internally every time a benchmark is called. It would be nice if there was a way to create two identical benchmark instances, for example by passing a seed to the environment as an argument.

Currently, I get:
```python
>>> import pickle
>>> import metaworld
>>> mt10_1, mt10_2 = metaworld.MT10(), metaworld.MT10()
>>> pickle.loads(mt10_1.train_tasks[0].data)['rand_vec']
array([-0.05115513,  0.63916668,  0.02      ,  0.07276333,  0.86412829,
        0.12647354])
pickle.loads(mt10_2.train_tasks[0].data)['rand_vec']
>>> pickle.loads(mt10_2.train_tasks[0].data)['rand_vec']
array([ 0.09087166,  0.69323066,  0.02      , -0.0617444 ,  0.81741752,
        0.14243739])
```

Would be useful if something like the following was possible:
```python
import numpy as np
import metaworld

seed = 10
mt10_1, mt10_2 = metaworld.MT10(seed=seed), metaworld.MT10(seed=seed)
rand_vec_1 = pickle.loads(mt10_1.train_tasks[0].data)['rand_vec']
rand_vec_2 = pickle.loads(mt10_2.train_tasks[0].data)['rand_vec']
np.testing.assert_equal(rand_vec_1, rand_vec_2)
```

avnishn, 2020-09-10:
Hey @hartikainen,

Thanks for bringing this feature request to our attention.

This is actually a similar but not quite duplicate request to #217. LMK if you disagree, but we are working on this. 

Granted we are more than happy to accept any new features through pull requests if you'd like to contribute one!

I'm going to leave this open for now, as to make sure it is included in the conversation w.r.t to #217. 

Thanks!
Avnish

hartikainen, 2020-09-10:
Thanks for quick the reply! #217 is so massive that it's not obvious how that relates to this feature request. Would you mind pointing me to the part of the code that helps towards this?

avnishn, 2020-09-10:
Oh sorry about that I linked the wrong issue. I meant #147 my b.

ryanjulian, 2020-09-10:
@hartikainen does pickling/unpicking and/or copy.deepcopy achieve this right now? It won't get you cross-experiment uniformity but it is a step. 

@avnishn is this FR tracked in an issue?

avnishn, 2020-09-10:
@ryanjulian the specific feature where we add this as a part of the constructor isn't tracked, although the general feature where we give the option to set the seed is tracked in #147.

closed 2021-04-15 by avnishn