Farama-Foundation/Metaworld #237: Correct set task for V2 Env

buoyancy99, 2020-10-14:
How can I successfully find task for v2 environments? All I found in ML1, ML45 etc are tasks with env_name ending with v1

ryanjulian, 2020-10-14:
@avnishn

ryanjulian, 2020-10-14:
@buoyancy99 thanks for filing this issue. I'm glad you've noticed there are changes afoot in the repo!

In short, we don't yet support constructing the benchmarks with "v2" environments because those environment versions are a work in progress [1]. We hope to finish them up and release them in the next month or two. We may add an experimental flag shortly which constructs the benchmarks with the "v2" environments, depending on when @avnishn gets around to it.

For your own experimental purposes, you are free to modify the `Benchmark` classes to use the v2 environments, but please don't report results from those environments as corresponding to the metaworld benchmark until we officially release them and update the benchmark. We have no plans to modify the `set_task` API for "v2" so your code probably won't change when we switch to "v2."

[1] The goal of the "v2" environments is to make metaworld more robustly trainable by many different algorithms, and to provide a clearer picture of relative performance. These are the same tasks as "v1", but may include different observation spaces, dynamics, and even reward functions, so you can't compare results from one algorithms "v1" to results from a different algorithm on "v2."

closed 2020-10-14 by ryanjulian