Farama-Foundation/Metaworld #272: env.render save images

haoyu-x, 2020-11-17:
Hi,
thanks for sharing this amazing project!

I'm trying to use a single task to do pixel-based RL.
But the env.render(mode = 'rgb_array') returns a top-down view only.
I'm wondering how to change the camera viewpoint? say, left-to-right.

Thanks!

ryanjulian, 2020-11-17:
@haydenshively knows a lot about this!

haoyu-x, 2020-11-18:
thanks @ryanjulian , I solved the issues. I looked into mujoco_env.py 	def viewer_setup(self):

haydenshively, 2020-11-18:
@haoyu-x Glad you figured it out! If you need additional camera views, let me know and I can try to add new cameras to the XML files. Also, I've only tested the existing camera views on some of the V2 environments, so let me know if you experience any issues

closed 2020-11-24 by avnishn

Tejas-Deo, 2023-05-08:
> @haoyu-x Glad you figured it out! If you need additional camera views, let me know and I can try to add new cameras to the XML files. Also, I've only tested the existing camera views on some of the V2 environments, so let me know if you experience any issues

Can you please share how did you get images from env.render()? Can you share the code changes that you did?
Thanks!