Farama-Foundation/Metaworld #277: How many 'shots' per task

danielamassiceti, 2020-11-20:
Hi there

I wondered if you could provide me with some details about the dataset:
1. How many 'shots' per task are used when adapting the model?
2. How many total samples/observations are in the dataset?
3. How many samples/observations are there per manipulation task in the dataset?

Thanks

ryanjulian, 2020-11-20:
@avnishn

avnishn, 2020-11-23:
Hi @danielamassiceti thanks for checking out Metaworld!

Though Metaworld could be thought of as a dataset, it is really actually a set of environments that an agent will interact with during its training process.

Let me try to answer your original questions to the best of my ability:

> How many 'shots' per task are used when adapting the model?

This is entirely up to you, there is no specific requirement by the simulator. You can use any amount of trajectories when doing meta-adaptation.

> How many total samples/observations are in the dataset?

Technically infinite. Since we're using a set of simulators, you can step the simulators an infinite number of times, although with Metaworld-v1, you'll need to train for about 20m timesteps for ML10 and MT10 and 100m timesteps for ML45 and MT50.

> How many samples/observations are there per manipulation task in the dataset?

Technically infinite. This is also up to the user, since when using a simulator, the amount of available data is dependent on how much you'd like to step the environment. The only limitation here is that each environment has a different max-path-length [Max path length means the maximum amount of times that a simulator can be stepped before it needs to be reset to its initial position, by calling env.reset()].

For each manipulation task, there are however 50 environment configurations (initial object and goal positions), which are found under MLXX.[train/test]_tasks, and while users aren't required to use all of them during training, many should be used during meta-adapt to display sensitivity of an agent's meta-adaptation to different environment configurations. These can be changed by making a call to env.set_task() using a corresponding task from MLXX.[train/test]_tasks.

Did this answer your questions? Please let me know/ask as many questions as you would like, here by @ing me, or on our slack community, which you can join via the sign-up link in our README.

Best,
@avnishn

ryanjulian, 2020-11-25:
I should note that though these are all true of the MetaWorld environments, the relevant hyperparameters (e.g. how many trajectories uses for adaptation) used for the results appearing in the MetaWorld publication are all in the appendix.

avnishn, 2020-11-29:
I'm going to go ahead and close this for now, but feel free to reopen if you have any additional questions @danielamassiceti!

closed 2020-11-29 by avnishn