Farama-Foundation/Metaworld #286: Gripper status not reflected in observation space?

mhauskn, 2020-12-10:
In my experiments with v1 and v2 tasks, as far as I can tell, the degree of open/closedness of the gripper isn't reflected in the 12-dimensional observation space. I imagine this makes many of the tasks significantly more challenging (e.g. trying to grip an object without observing your own gripper). I'd love to understand the intention behind this design choice.

avnishn, 2020-12-11:
Hi @mhauskn, thanks for checking out Metaworld!

Your observation is correct about the v1 environments, however, in the v2 environments, we updated them to have observation spaces that include a measurement of how open/closed the gripper is. 

We haven't updated any of the benchmarks to use the newest v2 environments, but we'll be doing that soon (in the next week or so) as apart of a large overhaul that we're completing with Metaworld!

Some things that we are going to update include:

- updated observation spaces to include measurements relevant to making all environments markovian.
- updated meshes for a new updated look
- updated reward functions that should be much easier to solve.

For now, if you want to toy with the v2 environments that we'll be releasing, you'll have to use them directly on the branch titled `new-reward-functions`. Keep in mind that we haven't merged all of the new reward functions as it's a WIP, but all of the updated ML10 and MT10 environments are on there.

Feel free to @me (@avnishn) if you have any more questions!

mhauskn, 2020-12-11:
Thank @avnishn - I'll give the ```new-reward-functions``` branch a try. Out of curiosity is there an ETA for v2 release?

avnishn, 2020-12-15:
@mhauskn expect it to be out in the next 2 weeks or so. We're very close to finishing it.

closed 2020-12-15 by avnishn