Farama-Foundation/Metaworld #297: How to save image of a camera with the scene rendered?

HeegerGao, 2021-01-12:
Hi, I am using keyboard control to collect demonstrations for imitation learning. I want to know if there is anyway to save image when I am rendering the scene (because I need to see the environment to perform keyboard control). 

I only find this method in the repo:
`env.sim.render(*resolution, mode='offscreen', camera_name=camera)[:,:,::-1]`

It works well when I am not rendering the scene. But when I use `env.render()` at the same time with this method, it will output an error:
```
  File "mjsim.pyx", line 156, in mujoco_py.cymj.MjSim.render
  File "mjsim.pyx", line 158, in mujoco_py.cymj.MjSim.render
  File "mjrendercontext.pyx", line 45, in mujoco_py.cymj.MjRenderContext.__init__
  File "mjrendercontext.pyx", line 113, in mujoco_py.cymj.MjRenderContext._setup_opengl_context
  File "opengl_context.pyx", line 130, in mujoco_py.cymj.OffscreenOpenGLContext.__init__
RuntimeError: Failed to initialize OpenGL
```

If I change the mode arguments to 'window', it outputs another error:
```
TypeError: 'NoneType' object is not subscriptable
```

Can you give me some advice? Thank you!

haydenshively, 2021-01-23:
Hi @HeegerGao,

Please try replacing the `render` function of `mujoco_env.py` with the following:
```
    def render(self, mode='human'):
        if mode == 'human':
            self._get_viewer(mode).render()
            return self.sim.render(
                *self._rgb_array_res,
                mode='offscreen',
            )[:, :, ::-1]
        elif mode == 'rgb_array':
            return self.sim.render(
                *self._rgb_array_res,
                mode='offscreen',
                camera_name='topview'
            )[:, :, ::-1]
        else:
            raise ValueError("mode can only be either 'human' or 'rgb_array'")
```

Then use `mode='human'` when you call `env.render`. This will cause mujoco to switch back and forth between the viewer (in which you should have keyboard control) and the rgb_array renderer. The function will return the pixels which you can then save using `cv2`. This works on my machine, but it seems to be pretty non-standard behaviour, so let me know if you have any trouble. Good luck!

avnishn, 2021-01-27:
I'm going to go ahead and close this for now @HeegerGao but please feel free to reopen it!

closed 2021-01-27 by avnishn