Farama-Foundation/Metaworld #303: Number of Experiments and Questions to Success Rate / Rewards

janbolle, 2021-01-15:
Dear all,

First of all, thank you very much for providing us with a nice benchmark like meta-world @tianheyu927 @zhanpenghe @ryanjulian!

I have a question about your experiment design:
1. How many experiments have you taken for ML1, ML10, and ML45?

Another thing is regarding the effect of Reward / Success Rate. The results you show are exciting.
For example, in the pick&place task, MAML gets a success rate of 0 but the highest reward.
2. I am wondering why that is the case? Can lead bad behavior to a high reward, but the success rate is low? If so, wouldn't we need to shape the reward function differently?

You described that you tried to make the reward function structurally fitting. But this seems odd to me. 
3. How would an agent learn the best policy if the success rate isn't somehow refracted in the reward function?

Any answer would be very appreciated!

[image]
[image]

ryanjulian, 2021-01-23:
@avnishn

avnishn, 2021-01-23:
Hi @janbolle,

Thanks for checking out metaworld! I am one of its maintainers and I am happy to answer your questions. 

1) meta rl and multi task rl algorithms are highly seed sensitive. You should run more than 1 seed of the same experiment on any of these benchmarks. I run 3 when I'm testing out ideas, and 5-10 when I run results. 

2) The success rates that you're seeing in the plots are reflective of our v1 environments. We have been working on v2 environments that include completely revamped reward functions. Specifically for pick-place, the new reward function we reduced the ability of the agent to reward hack (bad behavior leading to good rewards, like you said). 

You can check out those environments on our branch new-reward-functions. 

3) the reward function and success rates are based on physical quantities in the environment (such as position of the puck in pick and place). The new environments are written in such a way that the maximum reward is only achieved when we enter the successful state. Further more an environment can't observe its success rate. The success rate of an agent is computed by taking multiple episodes from that agent in an environment, and checking what fraction of those episodes entered a successful state. For pick and place it would mean how many episodes an agent successfully pick up the puck and place it at the goal location. If in 7 out of 10 episodes an agent was successful, then its success rate would be 70 percent. 

I hope this explanation explains why the success rate metric can't be directly incorporated into the reward. 

Have I answered all your questions? Please @ me, @avnishn , if you want to follow up, or join our slack community (sign up link in th e read me) for faster responses! 

Thanks!
@avnishn

avnishn, 2021-01-27:
I'm going to go ahead and close this for now, but please feel free to reopen @janbolle

closed 2021-01-27 by avnishn