Farama-Foundation/Metaworld #324: Inconsistent initialization of object positions

mtcrawshaw, 2021-02-23:
6 of the v1 environments (names listed in the output below) seem to have an issue where the initial object positions (elements 3 through 9 (exclusive) of the observation) for the very first episode of an environment instantiation is different than the initial object positions for the remaining episodes. I wrote a minimum working example and included it and the output below. This may seem like an insignificant problem, and it doesn't really matter in a lot of cases because the importance of the first episode is dwarfed as training gets longer and longer. However, it is crucial in some cases. In my training code, I included an option that creates a new environment instance at the beginning of each episode. This allows for training on MT50 with a much smaller memory requirement, since we only need to store one environment instance at a time instead of 50, so that I can train MT50 on my laptop. But in this case, every episode is the very first episode of an environment instantiation, so the goals will only ever be set to the initial (erroneous) value. I'm not sure how to fix this myself because I assume this has something to do with the way the Mujoco model is set up, but I haven't dug into that part of metaworld very much. Any help with this would be appreciated. Thanks!

```python3
import numpy as np
import random
import metaworld

EPISODE_LEN = 200
NUM_EPISODES = 10

def get_obj_pos(obs):
    return list(obs[3:9])

# Create benchmark.
benchmark = metaworld.MT50()
env_dict = benchmark.train_classes
tasks = benchmark.train_tasks
initial_obj_poses = {name: [] for name in env_dict.keys()}

# Execute rollout for each environment in benchmark.
for env_name, env_cls in env_dict.items():

    # Create environment and set task.
    env = env_cls()
    env_tasks = [t for t in tasks if t.env_name == env_name]
    env.set_task(random.choice(env_tasks))

    # Step through environment for a fixed number of episodes.
    for episode in range(NUM_EPISODES):
        
        # Reset environment and extract initial object position.
        obs = env.reset()
        initial_obj_pos = get_obj_pos(obs)
        initial_obj_poses[env_name].append(initial_obj_pos)

        # Environment steps.
        for step in range(EPISODE_LEN):
            a = env.action_space.sample()
            obs, reward, done, info = env.step(a)

# Display initial object positions and find environments with non-unique positions.
violating_envs = []
for env_name, task_initial_pos in initial_obj_poses.items():
    print("%s:\n[" % env_name)
    for obj_pos in task_initial_pos:
        print("    %s" % str(obj_pos))
    print("]\n")
    if len(np.unique(np.array(task_initial_pos), axis=0)) > 1:
        violating_envs.append(env_name)

# Print violating environments.
print("Violating environments: %s" % violating_envs)
```
and the resulting output:
```
reach-v1:
[
    [-0.004600083397474947, 0.6172984579395253, 0.019999999552965164, 0.0, 0.0, 0.0]
    [-0.004600083397474947, 0.6172984579395253, 0.019999999552965164, 0.0, 0.0, 0.0]
    [-0.004600083397474947, 0.6172984579395253, 0.019999999552965164, 0.0, 0.0, 0.0]
    [-0.004600083397474947, 0.6172984579395253, 0.019999999552965164, 0.0, 0.0, 0.0]
    [-0.004600083397474947, 0.6172984579395253, 0.019999999552965164, 0.0, 0.0, 0.0]
    [-0.004600083397474947, 0.6172984579395253, 0.019999999552965164, 0.0, 0.0, 0.0]
    [-0.004600083397474947, 0.6172984579395253, 0.019999999552965164, 0.0, 0.0, 0.0]
    [-0.004600083397474947, 0.6172984579395253, 0.019999999552965164, 0.0, 0.0, 0.0]
    [-0.004600083397474947, 0.6172984579395253, 0.019999999552965164, 0.0, 0.0, 0.0]
    [-0.004600083397474947, 0.6172984579395253, 0.019999999552965164, 0.0, 0.0, 0.0]
]

push-v1:
[... 662 lines cut ...]
    [0.04177085405220372, 0.7783651076990645, 0.1746032492641456, 0.0, 0.0, 0.0]
    [0.04177085405220372, 0.7783651076990645, 0.1746032492641456, 0.0, 0.0, 0.0]
    [0.04177085405220372, 0.7783651076990645, 0.1746032492641456, 0.0, 0.0, 0.0]
    [0.04177085405220372, 0.7783651076990645, 0.1746032492641456, 0.0, 0.0, 0.0]
    [0.04177085405220372, 0.7783651076990645, 0.1746032492641456, 0.0, 0.0, 0.0]
    [0.04177085405220372, 0.7783651076990645, 0.1746032492641456, 0.0, 0.0, 0.0]
    [0.04177085405220372, 0.7783651076990645, 0.1746032492641456, 0.0, 0.0, 0.0]
]

door-unlock-v1:
[
    [-0.025667779091990348, 0.7654568953587574, 0.07000001985681625, 0.0, 0.0, 0.0]
    [-0.025667779091990348, 0.7654568953587574, 0.07000001985681625, 0.0, 0.0, 0.0]
    [-0.025667779091990348, 0.7654568953587574, 0.07000001985681625, 0.0, 0.0, 0.0]
    [-0.025667779091990348, 0.7654568953587574, 0.07000001985681625, 0.0, 0.0, 0.0]
    [-0.025667779091990348, 0.7654568953587574, 0.07000001985681625, 0.0, 0.0, 0.0]
    [-0.025667779091990348, 0.7654568953587574, 0.07000001985681625, 0.0, 0.0, 0.0]
    [-0.025667779091990348, 0.7654568953587574, 0.07000001985681625, 0.0, 0.0, 0.0]
    [-0.025667779091990348, 0.7654568953587574, 0.07000001985681625, 0.0, 0.0, 0.0]
    [-0.025667779091990348, 0.7654568953587574, 0.07000001985681625, 0.0, 0.0, 0.0]
    [-0.025667779091990348, 0.7654568953587574, 0.07000001985681625, 0.0, 0.0, 0.0]
]

Violating environments: ['drawer-open-v1', 'window-open-v1', 'window-close-v1', 'faucet-open-v1', 'faucet-close-v1', 'lever-pull-v1', 'dial-turn-v1']
```

avnishn, 2021-02-25:
@mtcrawshaw Thanks a lot for bringing this bug up.

This is in fact a bug and not a design choice.

Essentially what has happened here is that we have logic to generate multiple random `(object_init_position, goal_position)` tuple, and in the case of the MT environments, we only generate 1 tuple. Each one of these tuples is what differentiates 2 tasks of the same environment.

The problem that you are facing is that there is an initial position specified by each of the mujoco files.
If you look at the initial position for all 6 environments you've listed, the object initial position is different only for the first task that is sampled, and the same for the rest of them. 

This is a pretty easy fix, and I'll be sure to have it out soon (less than a week)

Thanks for bringing this up,
@avnishn  

ps @krzentner can you look at this too, thanks!

avnishn, 2021-02-25:
Hmm upon investigating your bug what I initially said was the case is not actually the case @mtcrawshaw. 

You are right to say that it has to do with mujoco.
In the environment files sometimes we directly change the state of the simulator by adjusting the positions of objects.
It requires us to make a call to sim.step, that way the simulator is stepped forward and the states of objects are updated properly.

PR #328 (2021-02-26): Force state rand vec updates to update the sim
In some environments, the locations of objects are updated
via changes to the underlying simulator.

Those changes don't reflect in the simulator until the simulator
itself has been stepped (a call to sim.step()).

This commit forces sim.step when state rand vecs are changed by calling
reset inside set_task.

closes #324
commit d9a75c45 (2021-02-26): Force state rand vec updates to update the sim (#328)

* Force state rand vec updates to update the sim

In some environments, the locations of objects are updated
via changes to the underlying simulator.

Those changes don't reflect in the simulator until the simulator
itself has been stepped (a call to sim.step()).

This commit forces sim.step when state rand vecs are changed by calling
reset inside set_task.

closes #324
closed 2021-02-26 by avnishn