Farama-Foundation/Metaworld #332: Import metaworld Error

csufangyu, 2021-03-11:
when I try to use metaworld,I get a error.it is:
```
>>> import metaworld
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages/metaworld/__init__.py", line 20
    env_name: EnvName
            ^
SyntaxError: invalid syntax

```
could you help me?

avnishn, 2021-03-11:
@csufangyu I think you've installed the metaworld package incorrectly.

You should go ahead and do the installation procedure inside the readme. 

Cheers,
@avnishn

closed 2021-03-11 by avnishn

csufangyu, 2021-03-11:
hi，I do the installation according to the readme.and this is my procedure:
`(metaworld) fangyu@Robot-DGX-Station:~/robot_learning$ pip install git+https://github.com/rlworkgroup/metaworld.git@master#egg=metaworld
DEPRECATION: Python 3.5 reached the end of its life on September 13th, 2020. Please upgrade your Python as Python 3.5 is no longer maintained. pip 21.0 will drop support for Python 3.5 in January 2021. pip 21.0 will remove support for this functionality.
Looking in indexes: https://mirrors.aliyun.com/pypi/simple/
Collecting metaworld
  Cloning https://github.com/rlworkgroup/metaworld.git (to revision master) to /tmp/pip-install-vr6q4a4n/metaworld_148b3c6cef4e4f27adb155adeca4c0c6
  Running command git clone -q https://github.com/rlworkgroup/metaworld.git /tmp/pip-install-vr6q4a4n/metaworld_148b3c6cef4e4f27adb155adeca4c0c6
Requirement already satisfied: gym>=0.15.4 in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from metaworld) (0.17.3)
Requirement already satisfied: mujoco-py<2.1,>=2.0 in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from metaworld) (2.0.2.5)
Requirement already satisfied: numpy>=1.18 in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from metaworld) (1.18.5)
Requirement already satisfied: pyglet<=1.5.0,>=1.4.0 in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from gym>=0.15.4->metaworld) (1.5.0)
Requirement already satisfied: scipy in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from gym>=0.15.4->metaworld) (1.4.1)
Requirement already satisfied: cloudpickle<1.7.0,>=1.2.0 in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from gym>=0.15.4->metaworld) (1.6.0)
Requirement already satisfied: glfw>=1.4.0 in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from mujoco-py<2.1,>=2.0->metaworld) (2.1.0)
Requirement already satisfied: Cython>=0.27.2 in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from mujoco-py<2.1,>=2.0->metaworld) (0.29.22)
Requirement already satisfied: imageio>=2.1.2 in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from mujoco-py<2.1,>=2.0->metaworld) (2.9.0)
Requirement already satisfied: cffi>=1.10 in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from mujoco-py<2.1,>=2.0->metaworld) (1.14.5)
Requirement already satisfied: lockfile>=0.12.2 in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from mujoco-py<2.1,>=2.0->metaworld) (0.12.2)
Requirement already satisfied: pycparser in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from cffi>=1.10->mujoco-py<2.1,>=2.0->metaworld) (2.20)
Requirement already satisfied: pillow in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from imageio>=2.1.2->mujoco-py<2.1,>=2.0->metaworld) (7.2.0)
Requirement already satisfied: future in /home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages (from pyglet<=1.5.0,>=1.4.0->gym>=0.15.4->metaworld) (0.18.2)
Building wheels for collected packages: metaworld
  Building wheel for metaworld (setup.py) ... done
  Created wheel for metaworld: filename=metaworld-0.0.0-py3-none-any.whl size=84316144 sha256=54f96751c4d8997f9ddb3b7ec4f27ef83d7503ec112bb51811c0eaf9d832a1e1
  Stored in directory: /tmp/pip-ephem-wheel-cache-nebvuny4/wheels/04/48/e2/97bba5502231ba872082f8c74a9586d53a6bfa87fc4ef6e59e
Successfully built metaworld
Installing collected packages: metaworld
Successfully installed metaworld-0.0.0
(metaworld) fangyu@Robot-DGX-Station:~/robot_learning$ python
Python 3.5.6 |Anaconda, Inc.| (default, Aug 26 2018, 21:41:56)
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import metaworld
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/fangyu/anaconda3/envs/metaworld/lib/python3.5/site-packages/metaworld/__init__.py", line 20
    env_name: EnvName
            ^
SyntaxError: invalid syntax
>>> exit()
`

csufangyu, 2021-03-11:
ok，I have change the verson of python,and it is ok!

avnishn, 2021-03-11:
Oh I see the problem here. We support python 3.6 and greater.