Farama-Foundation/Metaworld #337: Inconsistent hand position initialization in ML45.

mtcrawshaw, 2021-03-18:
I've been running tests to verify assumptions about the variation in environments (goals, object positions, hand positions) and I've found an inconsistency in the initialization of hand positions for one environment, namely `peg-insert-side-v1`. In all other environments, every episode from the same environment starts with the same initial hand position (this actually was surprising to me, I was originally under the impression that these were varied. Am I mistaken here?). When running ML45 on the train environments, `peg-insert-side-v1` has several different possibilities for where the hand ends up at the beginning of the episode. I included an MWE below that demonstrates this problem. I have sort of narrowed down the source of this issue. During the call to `env.reset()`, everything in the process is consistent until `_reset_hand()` in `SawyerXYZEnv`. I don't quite understand the purpose of this function's implementation, but this function seems to iterate over simulation steps (in this case, 10 steps) and uses the final hand position after these simulation steps as the initial hand position for the episode. This simulation process seems to be the source of inconsistency between different episodes. I'm not sure why this is, as I'm not familiar with the underlying Mujoco models or code. This isn't a huge issue obviously but it was an inconsistency I noticed, and perhaps worth pointing out if there's a quick fix. MWE and results below. Note that you may have to change the seed in the MWE in order to produce the inconsistency.
```python3
import numpy as np
import random
import metaworld

EPISODE_LEN = 200
NUM_EPISODES = 10
SEED = 2

def get_hand_pos(obs):
    return list(obs[:3])

# Set random seed.
random.seed(SEED)
np.random.seed(SEED)

# Create benchmark.
benchmark = metaworld.ML45()
env_dict = benchmark.test_classes
tasks = benchmark.train_tasks
resample_tasks = True
initial_hand_poses = {name: [] for name in env_dict.keys()}

# Execute rollout for each environment in benchmark.
for env_name, env_cls in env_dict.items():

    # Create environment and set task.
    env = env_cls()
    env_tasks = [t for t in tasks if t.env_name == env_name]
    env.set_task(random.choice(env_tasks))

    # Step through environment for a fixed number of episodes.
    for episode in range(NUM_EPISODES):

        # Resample task, if necessary.
        if resample_tasks:
            env.set_task(random.choice(env_tasks))

        # Reset environment and extract initial hand position.
        obs = env.reset()
        initial_hand_pos = get_hand_pos(obs)
        initial_hand_poses[env_name].append(initial_hand_pos)

        # Environment steps.
        for step in range(EPISODE_LEN):
            a = env.action_space.sample()
            obs, reward, done, info = env.step(a)

# Display initial hand positions and find environments with non-unique positions.
violating_envs = []
for env_name, task_initial_pos in initial_hand_poses.items():

    print("%s:\n[" % env_name)
    for hand_pos in task_initial_pos:
        print("    %s" % str(hand_pos))
    print("]\n")
    rounded_hand_poses = np.array(task_initial_pos).round(decimals=3)
    if len(np.unique(rounded_hand_poses, axis=0)) > 1:
        violating_envs.append(env_name)

# Print violating environments.
print("Violating environments: %s" % violating_envs)
```
And results:
```
reach-v1:
[
    [-0.0326510705911765, 0.5148782583133935, 0.23688440075258166]
    [-0.0326510705911765, 0.5148782583133935, 0.23688440075258166]
[... 603 lines cut ...]
    [-0.034457081976547066, 0.5070419010890129, 0.23981724510317892]
    [-0.034457081976547066, 0.5070419010890129, 0.23981724510317892]
    [-0.034457081976547066, 0.5070419010890129, 0.23981724510317892]
    [-0.034457081976547066, 0.5070419010890129, 0.23981724510317892]
    [-0.034457081976547066, 0.5070419010890129, 0.23981724510317892]
    [-0.034457081976547066, 0.5070419010890129, 0.23981724510317892]
    [-0.034457081976547066, 0.5070419010890129, 0.23981724510317892]
]

dial-turn-v1:
[
    [-0.0344534110933349, 0.5070449138162056, 0.2398140532198081]
    [-0.0344534110933349, 0.5070449138162056, 0.2398140532198081]
    [-0.0344534110933349, 0.5070449138162056, 0.2398140532198081]
    [-0.0344534110933349, 0.5070449138162056, 0.2398140532198081]
    [-0.0344534110933349, 0.5070449138162056, 0.2398140532198081]
    [-0.0344534110933349, 0.5070449138162056, 0.2398140532198081]
    [-0.0344534110933349, 0.5070449138162056, 0.2398140532198081]
    [-0.0344534110933349, 0.5070449138162056, 0.2398140532198081]
    [-0.0344534110933349, 0.5070449138162056, 0.2398140532198081]
    [-0.0344534110933349, 0.5070449138162056, 0.2398140532198081]
]

Violating environments: ['peg-insert-side-v1']
```

avnishn, 2021-03-18:
@mtcrawshaw can you try reruning your experiments, but while using the v2 environments. We recently published them to master.

mtcrawshaw, 2021-03-23:
Yup, thanks for pointing this out. I updated to v2 environments and more of my tests failed actually. There are several environments that violate assumptions for consistency of goals, initial object positions, and initial hand positions for ML45 train/test. The assumptions that I made (which I believe are how the code is intended to work) for ML45 environments are:
- Goals should be resampled (i.e. not identical) between episodes
- Initial hand positions should be identical between episodes
- Initial object positions should be resampled (i.e. not identical) between episodes

However each one of these assumptions seems to be violated by several environments. Again, I haven't looked into why this might be. I'm particularly unfamiliar with the v2 environments. But as before, I have to assume this has something to do with the interaction between the Mujoco models and the python environment implementations. MWE below, and I omitted results this time for length.
```python3
import random
import pickle

import numpy as np
import metaworld
from metaworld import Task

SEED = 1
EPISODE_LEN = 200
NUM_EPISODES = 10
DECIMAL_PRECISION = 3

def get_goal(obs):
    return list(obs[36:39])

def get_hand_pos(obs):
    return list(obs[:3])

def get_obj_pos(obs):
    return list(obs[3:17])

def check_obs(env_dict, tasks, resample_tasks, add_observability):

    # Initialize obs info.
    goals = {name: [] for name in env_dict.keys()}
    hand_poses = {name: [] for name in env_dict.keys()}
    obj_poses = {name: [] for name in env_dict.keys()}

    # Execute rollout for each environment in benchmark.
    for env_name, env_cls in env_dict.items():

        # Create environment and set task.
        env = env_cls()
        env_tasks = [t for t in tasks if t.env_name == env_name]
        if add_observability:
            for i in range(len(env_tasks)):
                task = env_tasks[i]
                task_data = dict(pickle.loads(task.data))
                task_data["partially_observable"] = False
                env_tasks[i] = Task(env_name=task.env_name, data=pickle.dumps(task_data))
        env.set_task(random.choice(env_tasks))

        # Step through environment for a fixed number of episodes.
        for episode in range(NUM_EPISODES):

            # Resample task, if necessary.
            if resample_tasks:
                env.set_task(random.choice(env_tasks))

            # Reset environment and extract initial hand/object position and goal.
            obs = env.reset()
            goals[env_name].append(get_goal(obs))
            hand_poses[env_name].append(get_hand_pos(obs))
            obj_poses[env_name].append(get_obj_pos(obs))

            # Environment steps.
            for step in range(EPISODE_LEN):
                a = env.action_space.sample()
                obs, reward, done, info = env.step(a)

    return goals, hand_poses, obj_poses

# Set random seed.
random.seed(SEED)
np.random.seed(SEED)

# Create kwargs list for ML45_train and ML_45 test.
kwargs_list = []
benchmark = metaworld.ML45()
kwargs_list.append({"env_dict": benchmark.train_classes, "tasks": benchmark.train_tasks, "resample_tasks": True, "add_observability": True})
kwargs_list.append({"env_dict": benchmark.test_classes, "tasks": benchmark.test_tasks, "resample_tasks": True, "add_observability": True})

# Get list of goals, initial hand positions, and initial object positions for each task.
goals = {}
hand_poses = {}
obj_poses = {}
for kwargs in kwargs_list:
    benchmark_goals, benchmark_hand, benchmark_obj = check_obs(**kwargs)
    goals.update(benchmark_goals)
    hand_poses.update(benchmark_hand)
    obj_poses.update(benchmark_obj)

# Find environments that violate assumptions about observation info.
goal_violating_envs = []
hand_violating_envs = []
obj_violating_envs = []
for env_idx, env_name in enumerate(goals.keys()):

    # Check that goals aren't identical across episodes.
    task_goals = np.round(np.array(goals[env_name]), decimals=DECIMAL_PRECISION)
    if len(np.unique(task_goals, axis=0)) == 1:
        goal_violating_envs.append((env_idx, env_name))

    # Check that initial hand positions are identical across episodes.
    task_hand_poses = np.round(np.array(hand_poses[env_name]), decimals=DECIMAL_PRECISION)
    if len(np.unique(task_hand_poses, axis=0)) > 1:
        hand_violating_envs.append((env_idx, env_name))

    # Check that initial object positions aren't identical across episodes.
    task_obj_poses = np.round(np.array(obj_poses[env_name]), decimals=DECIMAL_PRECISION)
    if len(np.unique(task_obj_poses, axis=0)) == 1:
        obj_violating_envs.append((env_idx, env_name))

# Print violating environments.
print("Goal violating environments: %s" % goal_violating_envs)
print("Hand violating environments: %s" % hand_violating_envs)
print("Object violating environments: %s" % obj_violating_envs)
```

gunnxx, 2022-11-13:
I think it works as intended. `metaworld` has two types of variations, parametric variations (goal, initial hand position, initial object position) and non-parametric variations (eg. hammer and reach). I will call non-parametric variations as tasks.

To make my point clearer, I explain the observation space of the metaworld v2. The observation space is 39 for v2 which consists of `[current_obs(18), prev_obs(18), goal(3)]` where the number inside the bracket indicates size. The 18 elements of current observation and previous observation are composed of `[end_effector(4), object_1(7), object_2(7)]`. Not all tasks have all the components. If it only has one object in the environment, it will zeros out all the `object_2` observation (eg. `reach-v2`).

Different tasks has different types of parametric variations. As an example, `hammer-v2` parametric variations only differ in the hammer initial position. You can see it in the `SawyerHammerEnvV2.reset_model()` (in fact you can see which parametric variations each tasks have by seeing the `reset_model()` method).
```Python3
## SawyerHammerEnvV2
def reset_model(self):
    self._reset_hand()
  
    # Set position of box & nail (these are not randomized)
    self.sim.model.body_pos[self.model.body_name2id(
        'box'
    )] = np.array([0.24, 0.85, 0.0])
    # Update _target_pos
    self._target_pos = self._get_site_pos('goal')
  
    # Randomize hammer position
    self.hammer_init_pos = self._get_state_rand_vec() if self.random_init \
        else self.init_config['hammer_init_pos']
    self.nail_init_pos = self._get_site_pos('nailHead')
    self.obj_init_pos = self.hammer_init_pos.copy()
    self._set_hammer_xyz(self.hammer_init_pos)
  
    return self._get_obs()
```

All you have to look is where the output of `self._get_state_rand_vec()` puts its value. In this case it is put in `self.hammer_init_pos` which means the parametric variations correspond to the hammer initial position only. It might sounds silly because it will be easy tasks, no? However, the difficulty will come from generalizing to other environments (ie. learning to open window quickly given that you know how to manipulate hammer).

P.S. I don't look all the violating environments but you can check its `reset_model()` method.

reginald-mclean, 2023-02-01:
As Meta-World ownership has been transferred from the RLWorkGroup to [Farama Foundation](https://farama.org/), I am closing this issue. If there are any questions or requests for features please join our [Discord](https://discord.gg/PfR7a79FpQ)

closed 2023-02-01 by reginald-mclean