Farama-Foundation/Metaworld #342: Experiment parameters to reproduce PEARL results

janbolle, 2021-04-19:
Hello all,

I am trying to reproduce the PEARL results from your paper.
Can you guide me to the place where you configured the experiments?

I try to figure out the experiment parameters...
How many different task instances per task did you use for ML10 and ML50?

I really appreciate any help you can provide.

janbolle, 2021-04-23:
@ryanjulian, @avnishn maybe you could help me here?
In PEARL the latent_variable size is really important. Which value did you set here?
I also struggle with '# of samples / # of train steps per iteration | 22500 env steps / 4,000 training steps'
How do you get the # of samples? In PEARL there are multiple parameters as num_initial_steps, num_steps_prior, etc.
If you used the PEARL implementation from Kate Rakelly* I really would appreciate it if you could make the config file available.

* https://github.com/katerakelly/oyster

avnishn, 2021-04-23:
hi @janbolle 

I have been trying to tune a pearl agent for use on benchmarking metaworld-v2 as well.
I haven't had a great amount of success in trying.

Here is my launcher file for ML10:
https://github.com/rlworkgroup/garage/blob/avnish-new-metaworld-results-mt1/metaworld_launchers/ml10/pearl_metaworld_ml10.py

I use the PEARL implementation that we wrote in our sister repo, https://github.com/rlworkgroup/garage.

I've modified it on the branch I linked to include lots of training and evaluation statistics.

The training of the encoder is what really matters with PEARL. You'll want to pay attention to the batch size for your encoder (larger is generally better), latent size (not to small, not to big), and the KL loss, which is used for training the encoder (I log this quantity in my modified PEARL implementation). 

Again, I really haven't had much success in tuning our PEARL implementation to succeed on the ML benchmarks. If you want to take a crack at it that would be great.

Thanks!
@avnishn

janbolle, 2021-04-23:
@avnishn Thank you very much for your fast reply!
What does "haven't had a great amount of success in trying" exactly mean?
Do you have some charts you can share?

I definitely will share my parameters, if I can get something out of it.
Unfortunately, I wasn't aware of the PEARL garage implementation.
I added meta world to Kate's code and I'm planning to parallelize sampling of the environment to boost training time.
Maybe this is something which is also interesting for you.

avnishn, 2021-06-29:
please check out our V2 release on arxiv for the hyperparameters we officially used

closed 2021-06-29 by avnishn