Farama-Foundation/Metaworld #344: Baseline results on v2 benchmarks

ankeshanand, 2021-05-21:
Hi, 

Since the benchmarks are moving to v2 environments, I would imagine this would change the baseline results too. Have you tried running the baselines on v2 benchmarks, and if so could you share the results? This would make it much easier to run and compare algorithms on the new version.

krzentner, 2021-05-22:
We definitely intend to share results for v2 as soon as we've finished running and aggregating them. @avnishn has been working on that, and I expect we will release those results within the next couple of weeks.

GoingMyWay, 2021-06-01:
> We definitely intend to share results for v2 as soon as we've finished running and aggregating them. @avnishn has been working on that, and I expect we will release those results within the next couple of weeks.

Great!!!

YeeCY, 2021-06-08:
Waiting for an update

ankeshanand, 2021-06-16:
Hi @avnishn, Do you know if there's an update on v2 benchmark results?

avnishn, 2021-06-16:
We've had baselines available in our sister repo, garage, for a while now. 

I'm updating them with what I ran to get the v2 baseline results. 

You can track my progress here:

https://github.com/rlworkgroup/garage/pull/2287

The updates are stable, I'm just adding documentation and tests in order to pass our code merge-ability checks.

luckeciano, 2021-07-23:
Hello @avnishn,

Thanks for providing us your PR. Quick question: is it possible to also share the "progress.csv" files of the baseline runs? It would save a lot of time and compute resources for those who want to have the baseline curves (as we have for Figures 12~18). 

Thanks!

avnishn, 2021-07-23:
I'll have to get back to you on that @luckeciano. Although I'd recommend re-running your results for any benchmarks that you decide to use, because there might be algorithmic differences in the implementations of these algorithms that you use vs. the ones that I wrote.

reginald-mclean, 2023-02-02:
As Meta-World ownership has been transferred from the RLWorkGroup to [Farama Foundation](https://farama.org/), I am closing this issue. If there are any questions or requests for features please join our [Discord](https://discord.gg/PfR7a79FpQ)

closed 2023-02-02 by reginald-mclean