Farama-Foundation/Metaworld #346: Engineered reward for "sawyer_push_v2" envirnonment.

mch5048, 2021-06-29:
Hi, 

In the  `sawyer_push_v2`  environment, the reward for each step is computed as 
written [here](https://github.com/rlworkgroup/metaworld/blob/9e3863d3a3b7d3f0ab6f1b759bb156abbbb27a1a/metaworld/envs/mujoco/sawyer_xyz/v2/sawyer_push_v2.py#L165).

 What I am curious about is these lines of code,

`       
reward = 2 * object_grasped
if tcp_to_obj < 0.02 and tcp_opened > 0:
      reward += 1. + reward + 5. * in_place`

which seems to add  `object_grasped` twice.

Is this intentional or a typo? If intentional, what is the reason for this design choice?

Thanks in advance!

krzentner, 2021-06-30:
Typically, our reward functions are made up of logical terms computed using the `tolerance` function, then combined together using the Hamacher product, before being combined by a linear weighting of terms.

The code here is somewhat confusing because it gives the `object_grasped` term a weight of `3` in a surprising way.  The coefficients in this linear combination are found empirically, so if they work, so we're not changing them.

Having said that, this does look like a small discrepancy between the code and the reward function listed in the paper. We will update the paper to have the correct coefficient. Thanks for finding it!

mch5048, 2021-07-03:
Thank you for the clarification!

closed 2021-07-03 by mch5048