Farama-Foundation/Metaworld #349: May I ask a slightly silly question

TriBall3, 2021-08-05:
Hello！I am new to Mujoco and may ask a slightly silly question.
I followed your steps step by step, ran the code you provided, and found no results at last. I don't know where I did wrong. Could you please give me some advice?
May I ask if the source code you provide is just the underlying library code, similar to NUMPY, and we have to write the framework code of the final reinforcement learning?
[image]

krzentner, 2021-08-06:
This library only contains the environments (similar to OpenAI's Gym). If you want RL code as well, please take a look at [garage](https://github.com/rlworkgroup/garage), which we also maintain and has examples for Meta-World (the Meta-World results reported in the paper were run using [this branch](https://github.com/rlworkgroup/garage/tree/merge-mw-v2-examples)).

closed 2021-08-06 by krzentner

TriBall3, 2021-08-12:
Hello！I ran the program in Garage, but there was no visual code in it.

I changed the program here：

But the program reported an error

Can you tell me how to visualize the interface like in the paper?