Farama-Foundation/Metaworld #351: Mismatch of render() interface between MetaWorld and Gym

ottofabian, 2021-08-20:
Currently the render function of MujocoEnv does not take a `mode` argument for the `render()` function and instead exposes the `offscreen` flag. 
https://github.com/rlworkgroup/metaworld/blob/a0009ed9a208ff9864a5c1368c04c273bb20dd06/metaworld/envs/mujoco/mujoco_env.py#L121-L133
This is an issue when using the environments e.g. with any form of gym wrappers. As seen here https://github.com/openai/gym/blob/2853ce4797a4dadb1f8702c8e35143d9fbd43ea6/gym/core.py#L253-L254 the mode is passed as argument to the underlying environment per default. If render is not explicitly called as `render(False)`, we have `offscreen='human'` and the check for `not offscreen` will fail.

From what I can see the `offscreen` flag seems to be the same as `mode=human` and `mode=rgb_array` in the base gym interface. Is there any specific reason the `offscreen` flag is introduced over the default behavior?
If not, I think the `render()` function should not be overloaded this way to remain compatible with existing gym functionality. For the remaining kwargs I do not see any issues, as they fit the interface. 

Thanks for looking into this.

reginald-mclean, 2023-02-07:
This will could be fixed by changing all environments to use mujoco (and the gymnasium MujocoEnv)

csimo005, 2023-04-10:
Is there any reason in general why the mujoco base env is re-impelmented instead of inherited from mujoco_py?

reginald-mclean, 2023-06-21:
@csimo005 it is a heavily modified version of the mujoco base env. In a future release we are going to be using the mujoco base env from Gymnasium.

closed 2023-08-03 by reginald-mclean