Farama-Foundation/Metaworld #357: Reproducing Figure 11 and reporting success rate

rasoolfa, 2022-01-09:
Hi all and @avnishn,

I've been trying to reproduce results in Figure 11 of https://arxiv.org/pdf/1910.10897.pdf using  https://github.com/rlworkgroup/garage/blob/08492007d6e2d9ead9beb83a8a4247e52019ac7d/metaworld_examples/sac_metaworld.py and hyper-parameter reported in Table 3.

One thing which is not clear to me how the success rate is reported. I notice the env.step returns 'success' but want to verify here that is what reported in the paper. Here is the code the I use to report results ( random action is used for simplicity):

```
from metaworld.envs import ALL_V2_ENVIRONMENTS_GOAL_OBSERVABLE
env_cls = ALL_V2_ENVIRONMENTS_GOAL_OBSERVABLE['hammer-v2-goal-observable']
eval_env= env_cls(seed=0)
eval_env.seed(0)
avg_reward = 0 
success_rate = 0 
num_evals = 2

for _ in range(num_evals):
    obs = eval_env.reset()
    done = False
    stp = 0
    while not done and stp < eval_env.max_path_length:
        obs, reward, done, info = eval_env.step(eval_env.action_space.sample())
        avg_reward += reward
        stp += 1
        if 'success' in info:
            success_rate += info['success']
avg_reward /= num_evals
success_rate /= num_evals
```
Is this the right way to report the success rate?
Thanks for your help.
Rasool

rasoolfa, 2022-01-12:
I have to add, my results are way worse than reported results.  
Any help would be highly appreciated. Thanks.

avnishn, 2022-01-12:
Hi @rasoolfa,

sorry for the late response.

This is the correct way of computing success:
```
num_evals = 10
num_successful_eval_trajectories = 0

for _ in range(num_evals):
    obs = w.reset()
    done = False
    success_curr_time_step = False
    stp = 0
    while not done and stp < eval_env.max_path_length:
        obs, reward, done, info = eval_env.step(eval_env.action_space.sample())
        stp += 1
        if 'success' in info:
            success_curr_time_step = True
    num_successful_eval_trajectories += int(success_curr_time_step)

success_rate = num_successful_eval_trajectories/num_evals
```

The code hasn't shifted, or the environments, so its unlikely that a performance regression happened, unless there was a performance regression caused by one of the dependencies (e.g. an upgraded version of torch. I used 1.8)

Thanks,
@avnishn

rasoolfa, 2022-01-12:
Thanks Avnish.
So the value of 'success' in info, i.e. info = { 'success':val},  is not important then and should be ignored? 
Also should I use Table 3 as a reference for the hyper-parameters?

avnishn, 2022-01-12:
1) whoops sorry, I edited my answer. Gave you the wrong answer the first time. 
2) yes, and they should be the same as the hparams in the launcher that you linked. 
3)yes.

rasoolfa, 2022-01-13:
thanks again for your help.

avnishn, 2022-01-13:
No problem! I'm gonna go ahead and close this for now, but if you have more questions, I'd recommend joining our slack  community (link on the readme), where a lot of questions like these have been answered, but of course feel free to post here again if you'd like.

closed 2022-01-13 by avnishn

rasoolfa, 2022-01-13:
Thanks again @avnishn. One last thing, do you happen to have learning curves or logs files for these experiments (i.e. Figure 11)? I just want to compare as I still can't reproduce paper results.

krzentner, 2022-01-23:
To be clear, since the above conversation was unclear to me: In MetaWorld, an episode is considered successful if the `info['success']` _ever_ becomes `1.0` during that episode. SuccessRate therefore needs to be computed across many episodes to be meaningful.

AnukritiSinghh, 2024-01-12:
I wanted to know that success_rate is calculated in evaluation part? and that is what is reported in the paper?