Farama-Foundation/Metaworld #359: [BUG] ML10 Tasks incorrectly specified in env_dict.py

jhejna, 2022-01-31:
in the ML10_V2 specification in env_dict.py I believe the dictionary value corresponding to 'button-press-topdown-v2' should be SawyerButtonPressTopdownEnvV2 instead of SawyerButtonPressEnvV2. This way the name would match the environment and ML10 would parallel MT10.

See here:
https://github.com/rlworkgroup/metaworld/blob/master/metaworld/envs/mujoco/env_dict.py#L391

krzentner, 2022-03-18:
That certainly would make sense. Unfortunately, I believe this discrepancy is intentional.

reginald-mclean, 2023-02-02:
As Meta-World ownership has been transferred from the RLWorkGroup to [Farama Foundation](https://farama.org/), I am closing this issue but it will be fixed in a future version of Meta-World. If there are any questions or requests for features please join our [Discord](https://discord.gg/PfR7a79FpQ)

closed 2023-02-02 by reginald-mclean