Farama-Foundation/Metaworld #360: Error when running experiment with garage

ruffiann, 2022-03-10:
Hello! Thank you for your efforts in running and providing the Meta-World baselines! I meet a problem while running your baseline codes, and need some help.
I have installed garage and tried to run the example by `python -m garage.examples.torch.maml_trpo_metaworld_ml1_push.py` command according to the documentation instructions, but no matter which .py file I run, it reports the following error, how can I run the experiment correctly?
`2022-03-10 10:54:18 | [maml_trpo_metaworld_ml1_push] Logging to /mnt/hdd1/guojiaqi/githubcode/metaworld-master/data/local/experiment/maml_trpo_metaworld_ml1_push
/mnt/hdd1/guojiaqi/githubcode/garage-master/src/garage/experiment/deterministic.py:37: UserWarning: Enabeling deterministic mode in PyTorch can have a performance impact when using GPU.
  'Enabeling deterministic mode in PyTorch can have a performance '
Traceback (most recent call last):
  File "/home/guojiaqi/anaconda3/lib/python3.7/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/home/guojiaqi/anaconda3/lib/python3.7/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/mnt/hdd1/guojiaqi/githubcode/garage-master/src/garage/examples/torch/maml_trpo_metaworld_ml1_push.py", line 91, in <module>
    maml_trpo_metaworld_ml1_push()
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/mnt/hdd1/guojiaqi/githubcode/garage-master/src/garage/experiment/experiment.py", line 369, in __call__
    result = self.function(ctxt, **kwargs)
  File "/mnt/hdd1/guojiaqi/githubcode/garage-master/src/garage/examples/torch/maml_trpo_metaworld_ml1_push.py", line 71, in maml_trpo_metaworld_ml1_push
    n_workers=meta_batch_size)
  File "/mnt/hdd1/guojiaqi/githubcode/garage-master/src/garage/sampler/ray_sampler.py", line 62, in __init__
    ray.init(log_to_driver=False, ignore_reinit_error=True)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/ray/_private/client_mode_hook.py", line 105, in wrapper
    return func(*args, **kwargs)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/ray/worker.py", line 985, in init
    job_config=job_config)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/ray/worker.py", line 1587, in connect
    lambda worker_info: sys.path.insert(1, current_directory))
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/ray/worker.py", line 387, in run_function_on_all_workers
    pickled_function = pickle.dumps(function)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/ray/cloudpickle/cloudpickle_fast.py", line 73, in dumps
    cp.dump(obj)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/ray/cloudpickle/cloudpickle_fast.py", line 620, in dump
    return Pickler.dump(self, obj)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/ray/pickle5_files/pickle5/pickle.py", line 485, in dump
    self.save(obj)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/ray/pickle5_files/pickle5/pickle.py", line 601, in save
    self.save_reduce(obj=obj, *rv)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/ray/pickle5_files/pickle5/pickle.py", line 689, in save_reduce
    save(func)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/ray/pickle5_files/pickle5/pickle.py", line 601, in save
    self.save_reduce(obj=obj, *rv)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/ray/pickle5_files/pickle5/pickle.py", line 689, in save_reduce
    save(func)
  File "/home/guojiaqi/anaconda3/lib/python3.7/site-packages/ray/pickle5_files/pickle5/pickle.py", line 576, in save
    rv = reduce(self.proto)
TypeError: can't pickle function objects
`

I asked other partners who have also used metaworld and he said he hadn't encountered such problems. But he is using an older version of metaworld, so I guess it could be a version problem? Where can I get an old version of metaworld or garaged?
I would appreciate it if you can provide some help!

krzentner, 2022-03-11:
It's difficult to tell just from the backtrace, but it looks like ray's cloudpickle is for some reason just using pickle, and then failing when it can't pickle a function (which is why cloudpickle is used in the first place).
What's especially weird is the error looks like it's in `ray.init`, which we don't pass anything especially interesting into.
What `ray` version do you have installed?
You can probably also work around this by not using the ray sampler in garage, although that does involve modifying the experiment files.

ruffiann, 2022-03-12:
The ray version I have installed is 1.11.0. I tried to lower the version to 0.8.7. but it still gives me this error. I also tried changing the metaworld version, but that didn't work either.  Maybe there is something else wrong with the configuration on my server.....  :(

avnishn, 2022-03-12:
Hmmm that's certainly possible. Did you also use the baselines from the pr branch that I linked. 

We haven't had any time to do garage maintenance for a while, so the baselines that I used when I ran the paper are sitting on a pr, and aren't the same as the ones on master. Maybe try those. You can find a link to the pr in another one of the open issues on this repo.

ruffiann, 2022-03-17:
a good news! I rebuilt a conda virtual env and reinstalled metaworld and garage in it. the problem was solved!

avnishn, 2022-03-17:
Awesome! Gonna close this now.

closed 2022-03-17 by avnishn