Farama-Foundation/Metaworld #363: How to Load and Use a Trained Policy?

ruffiann, 2022-03-18:
Thanks for your great project and contribution! I'm  so sorry to bother you again, but can you tell me how to load and use the trained policy?  I have successfully run the ml1 task and got the training results.  

I am trying to use the trained policy and visualize the results, according to the instructions in the linked document.[https://garage.readthedocs.io/en/latest/user/reuse_garage_policy.html#applying-the-policy-to-an-environment](url)

but I faid. My code is here:
`# Load the policy
from garage.experiment import Snapshotter
#import tensorflow as tf # optional, only for TensorFlow as we need a tf.Session

snapshotter = Snapshotter()

#with tf.compat.v1.Session(): # optional, only for TensorFlow
#    data = snapshotter.load('path/to/snapshot/dir')

data = snapshotter.load('data/local/experiment/maml_ppo_metaworld_ml1_push/')
policy = data['algo'].policy

# You can also access other components of the experiment
env = data['env']
steps, max_steps = 0, 150
done = False
obs = env.reset()  # The initial observation
policy.reset()

while steps < max_steps and not done:
    obs, rew, done, _ = env.step(policy.get_action(obs))
    env.render()  # Render the environment to see what's going on (optional)
    steps += 1

env.close()`

Error message is here:
`Traceback (most recent call last):
  File "read_policy.py", line 30, in <module>
    obs, rew, done, _ = env.step(policy.get_action(obs))
  File "/home/guojiaqi/.local/lib/python3.7/site-packages/garage/torch/policies/stochastic_policy.py", line 47, in get_action
    observation = as_torch(observation)
  File "/home/guojiaqi/.local/lib/python3.7/site-packages/garage/torch/_functions.py", line 146, in as_torch
    return torch.as_tensor(array).float().to(global_device())
TypeError: can't convert np.ndarray of type numpy.object_. The only supported types are: float64, float32, float16, complex64, complex128, int64, int32, int16, int8, uint8, and bool.`

I tried to change the format of obs, but I couldn't find the right way. Could you please provide a code file that can loads and uses the trained policy and can visualize the results? 

I would appreciate it if you can provide some help!

ruffiann, 2022-03-18:
I tried the code in another question, but still had problems. :(

the code is here:
`from garage.experiment import Snapshotter`
`import metaworld`
`import random`
`snapshotter = Snapshotter()`
`data = snapshotter.load('data/local/experiment/maml_ppo_metaworld_ml1_push_10epoch/')`
`policy = data['algo'].policy`
`env = data['env']`
`print('env', env)`
`print("Data dictionary",data)`
`from garage import rollout`
`path = rollout(env[0], policy, animated=True)`
`print(path)`

Error message is here:
`Traceback (most recent call last):`
`  File "read_policy.py", line 57, in <module>`
`    path = rollout(env[0], policy, animated=True)`
`TypeError: 'TaskNameWrapper' object does not support indexing`

krzentner, 2022-05-04:
If you're using the `TaskNameWrapper` like this, you should use `env` instead of `env[0]`.

closed 2022-05-04 by krzentner