Farama-Foundation/Metaworld #364: Seeding not working for v2 environments

buoyancy99, 2022-03-27:
I am trying to get deterministic behavior for the same seed> However, running the following piece of code prints about different result every time.

`
import metaworld
ml10 = metaworld.ML10()
cls_name = 'reach-v2'
env = ml10.train_classes[cls_name]()
task = [task for task in ml10.train_tasks if task.env_name == cls_name][0]
env.set_task(task)
print(env.obj_init_pos)
`

This is also the case for individual environments

`
from metaworld.envs import ALL_V2_ENVIRONMENTS_GOAL_HIDDEN
ReachEnv = ALL_V2_ENVIRONMENTS_GOAL_HIDDEN['reach-v2-goal-hidden']
env = ReachEnv(seed=0)
env.reset()   # doesn't matter if reset here. seeding not working in either case
print(env.obj_init_pos)
`

This could be an issue of
[line](https://github.com/rlworkgroup/metaworld/blob/29fe5d6d95cf9ad86f63eac38db8c0aef3837994/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py#L473)

ottofabian, 2022-05-24:
I have reported the same issue some time ago here #350. The issue is indeed using the global RNG.

krzentner, 2022-05-26:
Unfortunately this isn't quite as simple as just changing that one line, but there is now a PR (#370) that should address this use case. It also has a test that shows how to use it and I believe verifies that everything works as expected.

reginald-mclean, 2023-02-02:
I cannot replicate this issue, so I think PR (#370) has fixed this. As Meta-World ownership has been transferred from the RLWorkGroup to [Farama Foundation](https://farama.org/), I am closing this issue but it will be fixed in a future version of Meta-World. If there are any questions or requests for features please join our [Discord](https://discord.gg/PfR7a79FpQ)

closed 2023-02-02 by reginald-mclean