Farama-Foundation/Metaworld #366: ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject

Zhennan-Wu, 2022-04-13:
I tried to run the demo code
`
import metaworld
import random

print(metaworld.ML1.ENV_NAMES)  # Check out the available environments

ml1 = metaworld.ML1('pick-place-v1') # Construct the benchmark, sampling tasks

env = ml1.train_classes['pick-place-v1']()  # Create an environment with task `pick_place`
task = random.choice(ml1.train_tasks)
env.set_task(task)  # Set task

obs = env.reset()  # Reset environment
a = env.action_space.sample()  # Sample an action
obs, reward, done, info = env.step(a)  # Step the environoment with the sampled random action
`
but got the following error

Traceback (most recent call last):
  File "metaworld_test.py", line 1, in <module>
    import metaworld
  File "/home/leo/miniconda3/envs/metaworld/lib/python3.8/site-packages/metaworld/__init__.py", line 7, in <module>
    import metaworld.envs.mujoco.env_dict as _env_dict
  File "/home/leo/miniconda3/envs/metaworld/lib/python3.8/site-packages/metaworld/envs/__init__.py", line 1, in <module>
    from metaworld.envs.mujoco.env_dict import (ALL_V2_ENVIRONMENTS_GOAL_HIDDEN,
  File "/home/leo/miniconda3/envs/metaworld/lib/python3.8/site-packages/metaworld/envs/mujoco/env_dict.py", line 6, in <module>
    from metaworld.envs.mujoco.sawyer_xyz.v1 import (
  File "/home/leo/miniconda3/envs/metaworld/lib/python3.8/site-packages/metaworld/envs/mujoco/sawyer_xyz/v1/__init__.py", line 1, in <module>
    from metaworld.envs.mujoco.sawyer_xyz.v1.sawyer_assembly_peg import SawyerNutAssemblyEnv
  File "/home/leo/miniconda3/envs/metaworld/lib/python3.8/site-packages/metaworld/envs/mujoco/sawyer_xyz/v1/sawyer_assembly_peg.py", line 5, in <module>
    from metaworld.envs.mujoco.sawyer_xyz.sawyer_xyz_env import SawyerXYZEnv, _assert_task_is_set
  File "/home/leo/miniconda3/envs/metaworld/lib/python3.8/site-packages/metaworld/envs/mujoco/sawyer_xyz/sawyer_xyz_env.py", line 7, in <module>
    import mujoco_py
  File "/home/leo/miniconda3/envs/metaworld/lib/python3.8/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/leo/miniconda3/envs/metaworld/lib/python3.8/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/leo/miniconda3/envs/metaworld/lib/python3.8/site-packages/mujoco_py/builder.py", line 106, in load_cython_ext
    mod = load_dynamic_ext('cymj', cext_so_path)
  File "/home/leo/miniconda3/envs/metaworld/lib/python3.8/site-packages/mujoco_py/builder.py", line 130, in load_dynamic_ext
    return loader.load_module()
  File "cymj.pyx", line 1, in init mujoco_py.cymj
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject

Zhennan-Wu, 2022-04-13:
I installed mujoco first and ran its demo code, which went smoothly. But after I installed metaworld and encountered the error message above, I tried to reran the mujoco demo code, but it showed the same error.

Could someone help?

Thanks!

Zhennan-Wu, 2022-04-13:
I solved the problem. The reason is that I have both mujoco200 and mujoco 210 and the installation use the wrong cache.

closed 2022-04-13 by Zhennan-Wu

krzentner, 2022-04-14:
If anyone else encounters this error (or similar ones), I've also sometimes been able to resolve it by force installing an older version of numpy. Fortunately it doesn't seem to be a problem with mujoco_py 2.1.